cesanta/elk #2: AVR support

sfranzyshen, 2019-10-07:
are there any future plans to release the binary blob for the AVR platform?

sfranzyshen, 2019-10-07:
I have been informed that this is only a development pre-release ... and to expect a release in 2 weeks that will include additional platform support ... and new features :)

cpq, 2019-10-08:
This is still a pre-release.
0.0.3 just has been pushed with esp8266 and avr build. README partially updated - there was a breaking API change in elk.h. Please refer to elk.h as a source of truth.

For FFI, you need to know function address, and call `let f = ffi('signature@address');` from within JS. You can either use the `js_ffi()` C macro to obtain the address, or scan the firmware's ELF file to get the address of a function you'd like to FFI.

Still a WIP. If you manage to try this on AVR / esp8266, your feedback would be appreciated.

Closing this for now.

closed 2019-10-08 by cpq