cesanta/elk #4: Fails to compile under the Arduino IDE

sfranzyshen, 2019-10-08:
LINUX_DIST="Linux Mint 18.3 Sylvia MATE 32-bit"
KERNEL=Linux computer 4.10.0-38-generic i686 i686 i686 GNU/Linux

ARDUINO IDE: 1.8.10 (32bit)
Elk Library: release 0.0.3 (Installed manually into Arduino IDE library folder)
ESP8266 BOARD: 2.5.2 (Installed with board manager from Arduino IDE)

TEST CODE: (ArduinoBlink.ino)
```

#include "elk.h"  // Sketch -> Add File -> elk.h and elk.c

extern "C" void myDelay(int milli) {
  delay(milli);
}
extern "C" void myWrite(int pin, int val) {
  digitalWrite(pin, val);
}

void setup() {
  void *mem = malloc(500);
  struct js *js = js_create(mem, 500);
  js_ffi(js, myDelay, "vi");
  js_ffi(js, myWrite, "vii");
  js_eval(js,
          "while (1) { "
          "  myWrite(13, 0); "
          "  myDelay(100); "
          "  myWrite(13, 1); "
          "  myDelay(100); "
          "}",
          -1);
}

void loop() {
  delay(1000);
}

```
ESP8266 OUTPUT:
```
/home/name/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: ArduinoBlink.ino.elf section `.text1' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Generic ESP8266 Module.

```
NOTES:
Generic ESP8266 4MB (1MB spiffs)

AVR OUTPUT:
```
/home/name/Arduino/libraries/elk/src/atmega328p/libelk.a: In function `js_object_first_prop':
/Users/lsm/src/cpqelk/elk.c:517: undefined reference to `__mulhi3'
/home/name/Arduino/libraries/elk/src/atmega328p/libelk.a: In function `js_object_key':
/Users/lsm/src/cpqelk/elk.c:529: undefined reference to `__mulhi3'
/home/name/Arduino/libraries/elk/src/atmega328p/libelk.a: In function `js_ref':
/Users/lsm/src/cpqelk/elk.c:787: undefined reference to `__mulhi3'
/home/name/Arduino/libraries/elk/src/atmega328p/libelk.a: In function `js_get_prop':
/Users/lsm/src/cpqelk/elk.c:802: undefined reference to `__mulhi3'
/home/name/Arduino/libraries/elk/src/atmega328p/libelk.a: In function `js_object_first_prop':
/Users/lsm/src/cpqelk/elk.c:517: undefined reference to `__mulhi3'
/home/name/Arduino/libraries/elk/src/atmega328p/libelk.a:/Users/lsm/src/cpqelk/elk.c:657: more undefined references to `__mulhi3' follow
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino/Genuino Uno.

```
NOTES:

I had to rename avr to atmega328p in the src folder ...

sfranzyshen, 2019-10-08:
another weird thing happens ... I get a warning about multiple libraries were found ... when there isn't! the right version gets choose ...
```
 
Multiple libraries were found for "elk.h"
 Used: /home/name/Arduino/libraries/elk 
```

sfranzyshen, 2019-10-08:
LINUX_DIST="Linux Mint 18.3 Sylvia MATE 32-bit"
KERNEL=Linux computer 4.10.0-38-generic i686 i686 i686 GNU/Linux

ARDUINO IDE: 1.8.10 (32bit)
Elk Library: release 0.0.3 (Installed manually into Arduino IDE library folder)
ESP8266 BOARD: 2.6.0-dev (Installed manually into Arduino IDE user folder)

OUTPUT:
```
Multiple libraries were found for "elk.h"
 Used: /home/name/Arduino/libraries/elk
/home/name/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /tmp/arduino_build_110659/ArduinoBlink.ino.elf section `.text1' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
Using library elk at version 0.0.3 in folder: /home/name/Arduino/libraries/elk 
exit status 1
Error compiling for board Generic ESP8266 Module.
```
NOTES:
same toolchain used by platformio ...

QUESTIONS:
What toolchain was used to produce the library dot_a file??

cpq, 2019-10-10:
Arduino 1.8.10, with the latest board support for esp8266 and esp32.
Also, AVR GCC from Arduino 1.8.10 was used to produce AVR .a file.
Literally, 

```
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc -W -Wall -Wextra -pedantic -fstrict-aliasing -Os -g -c elk.c -o libelk.a
```

sfranzyshen, 2019-10-10:
I'm not clear on how things are loaded and not loaded into ram or flash ... but maybe somehow the libelk.a file is being mapped to ram ... maybe -DICACHE_FLASH added to the build could help?

sfranzyshen, 2019-10-10:
BTW I was successful building for the esp32 platform using the same setup ...

sfranzyshen, 2019-10-10:
and perhaps adding -mmcu=atmega328p to the avr build ...

cpq, 2019-10-11:
Added `-mmcu=atmega328p`, pushed 0.0.4. Adding `-DICACHE_FLASH` does not appear to make any difference on the output binary file. We'll take a look on how to pin functions on flash.

sfranzyshen, 2019-10-11:
LINUX_DIST="Linux Mint 18.3 Sylvia MATE 32-bit"
KERNEL=Linux computer 4.10.0-38-generic i686 i686 i686 GNU/Linux

ARDUINO IDE: 1.8.10 (32bit)
Elk Library: release 0.0.4 (Installed manually into Arduino IDE library folder)
ESP8266 BOARD: 2.5.2 (Installed with board manager from Arduino IDE)

TEST CODE: (ArduinoBlink.ino)

```
#include <elk.h>
//#define LED_BUILTIN 2

extern "C" void myDelay(int milli) {
  delay(milli);
}
extern "C" void myWrite(int pin, int val) {
  digitalWrite(pin, val);
}

void setup() {
  Serial.begin(115200);
  Serial.println("Starting");
  pinMode(LED_BUILTIN, OUTPUT);
  void *mem = malloc(500);
  struct js *js = js_create(mem, 500);
  Serial.println("VM ready");
  js_ffi(js, myDelay, "vi");
  js_ffi(js, myWrite, "vii");
  Serial.println("Functions mapped, evaluating ...");

  js_eval(js,
          "while (1) { "
          "  myWrite(LED_BUILTIN, 0); "
          "  myDelay(100); "
          "  myWrite(LED_BUILTIN, 1); "
          "  myDelay(100); "
          "}",
          -1);
  Serial.println("Evaluation returned ...");
}

void loop() {
  delay(1000);
}
```
ESP8266 OUTPUT:

```
Linking everything together...
/home/name/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-gcc -CC -E -P -DVTABLES_IN_FLASH /home/name/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/ld/eagle.app.v6.common.ld.h -o /tmp/arduino_build_589665/local.eagle.app.v6.common.ld
/home/name/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-gcc -fno-exceptions -Wl,-Map -Wl,/tmp/arduino_build_589665/ArduinoBlink.ino.map -g -w -Os -nostdlib -Wl,--no-check-sections -u app_entry -u _printf_float -u _scanf_float -Wl,-static -L/home/name/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/lib -L/home/name/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/lib/NONOSDK221 -L/home/name/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/ld -L/home/name/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/libc/xtensa-lx106-elf/lib -Teagle.flash.4m.ld -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read -o /tmp/arduino_build_589665/ArduinoBlink.ino.elf -Wl,--start-group /tmp/arduino_build_589665/sketch/ArduinoBlink.ino.cpp.o /home/name/Arduino/libraries/elk/src/esp8266/libelk.a /tmp/arduino_cache_816946/core/core_a3d205a9305a25a2e3ad9b96c1709807.a -lhal -lphy -lpp -lnet80211 -llwip2-536-feat -lwpa -lcrypto -lmain -lwps -lbearssl -laxtls -lespnow -lsmartconfig -lairkiss -lwpa2 -lstdc++ -lm -lc -lgcc -Wl,--end-group -L/tmp/arduino_build_589665
Multiple libraries were found for "elk.h"
 Used: /home/name/Arduino/libraries/elk
/home/name/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /tmp/arduino_build_589665/ArduinoBlink.ino.elf section `.text1' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
Using library elk at version 0.0.4 in folder: /home/name/Arduino/libraries/elk 
exit status 1
Error compiling for board Generic ESP8266 Module.

```
ESP8266 NOTES:
Generic ESP8266 4MB (no spiffs)

AVR OUTPUT:

```
Linking everything together...
/home/name/arduino-1.8.10/hardware/tools/avr/bin/avr-gcc -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p -o /tmp/arduino_build_589665/ArduinoBlink.ino.elf /tmp/arduino_build_589665/sketch/ArduinoBlink.ino.cpp.o /home/name/Arduino/libraries/elk/src/atmega328p/libelk.a /tmp/arduino_build_589665/../arduino_cache_816946/core/core_arduino_avr_nano_cpu_atmega328old_ec06627f7ed5ce24080a13662272f0fb.a -L/tmp/arduino_build_589665 -lm
/home/name/arduino-1.8.10/hardware/tools/avr/bin/avr-objcopy -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 /tmp/arduino_build_589665/ArduinoBlink.ino.elf /tmp/arduino_build_589665/ArduinoBlink.ino.eep
[... 6 lines cut ...]
Global variables use 1034 bytes (50%) of dynamic memory, leaving 1014 bytes for local variables. Maximum is 2048 bytes.

```
AVR NOTES:
I had to rename avr to atmega328p in the src folder ... It compiled but locks-up at js_create()  

ESP32 OUTPUT:

```
Linking everything together...
/home/name/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-gcc -nostdlib -L/home/name/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/lib -L/home/name/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/ld -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld -T esp32.rom.spiram_incompatible_fns.ld -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority -u __cxa_guard_dummy -u __cxx_fatal_exception -Wl,--start-group /tmp/arduino_build_589665/sketch/ArduinoBlink.ino.cpp.o /home/name/Arduino/libraries/elk/src/esp32/libelk.a /tmp/arduino_cache_816946/core/core_d3ae8402242720f190268245f30eba2e.a -lgcc -lesp32 -lphy -lesp_http_client -lmbedtls -lrtc -lesp_http_server -lbtdm_app -lspiffs -lbootloader_support -lmdns -lnvs_flash -lfatfs -lpp -lnet80211 -ljsmn -lface_detection -llibsodium -lvfs -ldl_lib -llog -lfreertos -lcxx -lsmartconfig_ack -lxtensa-debug-module -lheap -ltcpip_adapter -lmqtt -lulp -lfd -lfb_gfx -lnghttp -lprotocomm -lsmartconfig -lm -lethernet -limage_util -lc_nano -lsoc -ltcp_transport -lc -lmicro-ecc -lface_recognition -ljson -lwpa_ [... cut ...]
Using library elk at version 0.0.4 in folder: /home/name/Arduino/libraries/elk 
/home/name/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-size -A /tmp/arduino_build_589665/ArduinoBlink.ino.elf
Sketch uses 243601 bytes (18%) of program storage space. Maximum is 1310720 bytes.
Global variables use 15500 bytes (4%) of dynamic memory, leaving 312180 bytes for local variables. Maximum is 327680 bytes.

```

ESP32 NOTES:
things build ... but I do not have a esp32 with me to do physical test ...

sfranzyshen, 2019-10-11:
I was successful at building an earlier version of Elk (mjs3) as a dot_a library file ... and included it successfully as a pre-compiled library into the blink example for the esp8266 ... Here is how I did it ...

1) I edited the mjs3 library library.properties file and added dot_a_linkage=true to the end ... (/home/name/Arduino/libraries/mjs3/library.properties)
2) I built the blink example ...
3) After the build was complete I navigated to the build folder (In Linux it's located in /tmp/arduino_build_xxxxxx folder) then I copy the dot_a file from under library folder (libraries/mjs3.a) to the arduino library folder for the mjs3 library (/home/name/Arduino/libraries/mjs3/src/esp8266/libmjs3.a)
4) I edited the mjs3 library library.properties file again and removed dot_a_linkage=true and added precompiled=true and ldflags=-lmjs3 to the end of the file ...
5)  I build the blink example ... this time it links to the dot_a file successfully and runs ...

I removed the mjs3.c to confirm it was using the dot_a file to build ... i'm going to compare how the map files compare to elk ... i'll update as soon as i do ...

sfranzyshen, 2019-10-12:
it appears that this method works for avr platforms too ...

sfranzyshen, 2019-10-12:
@cpq :  tested avr build for v0.0.5 ... same setup targeting Arduino Nano (atmega328p)

```
Linking everything together...
/home/name/arduino-1.8.10/hardware/tools/avr/bin/avr-gcc -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p -o /tmp/arduino_build_956803/ArduinoBlink.ino.elf /tmp/arduino_build_956803/sketch/ArduinoBlink.ino.cpp.o /home/name/Arduino/libraries/elk/src/atmega328p/libelk.a /tmp/arduino_build_956803/../arduino_cache_120855/core/core_arduino_avr_nano_cpu_atmega328old_ec06627f7ed5ce24080a13662272f0fb.a -L/tmp/arduino_build_956803 -lm
/home/name/arduino-1.8.10/hardware/tools/avr/bin/avr-objcopy -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 /tmp/arduino_build_956803/ArduinoBlink.ino.elf /tmp/arduino_build_956803/ArduinoBlink.ino.eep
/home/name/arduino-1.8.10/hardware/tools/avr/bin/avr-objcopy -O ihex -R .eeprom /tmp/arduino_build_956803/ArduinoBlink.ino.elf /tmp/arduino_build_956803/ArduinoBlink.ino.hex
Multiple libraries were found for "elk.h"
 Used: /home/name/Arduino/libraries/elk
Using library elk at version 0.0.4 in folder: /home/name/Arduino/libraries/elk 
/home/name/arduino-1.8.10/hardware/tools/avr/bin/avr-size -A /tmp/arduino_build_956803/ArduinoBlink.ino.elf
Sketch uses 27262 bytes (88%) of program storage space. Maximum is 30720 bytes.
Global variables use 1042 bytes (50%) of dynamic memory, leaving 1006 bytes for local variables. Maximum is 2048 bytes.

```
builds but locks up when calling js_create() ...
```
  void *mem = malloc(500);
  Serial.println("before");
  struct js *js = js_create(mem, 500);
  Serial.println("VM ready");
```

cpq, 2019-11-04:
Pushed 0.0.6
The issue was a big VM size (it preallocated big buffers for stacks, objects, etc).
Reduced to ~400 bytes. Thus, give 500 bytes or more when calling `js_create`.

cpq, 2019-11-08:
Closing - please reopen if required

closed 2019-11-08 by cpq

sfranzyshen, 2019-11-09:
Same setup as above ...
esp8266 - Still not compiling ... (same issue as above)
avr - compiles, but does not perform js_eval() ... just returns ...