cesanta/elk #7: ESP8266 Fails to compile under the Arduino IDE

sfranzyshen, 2019-11-10:
[continued from issue #4](https://github.com/cesanta/elk/issues/4#issue-504223691)

LINUX_DIST="Linux Mint 18.3 Sylvia MATE 32-bit"
KERNEL=Linux computer 4.10.0-38-generic i686 i686 i686 GNU/Linux

ARDUINO IDE: 1.8.10 (32bit)
Elk Library: release 0.0.7 (Installed manually into Arduino IDE library folder)
ESP8266 BOARD: 2.5.2 (Installed with board manager from Arduino IDE)

`ArduinoBlink.ino.elf section `.text1' will not fit in region `iram1_0_seg'
`

it seams that the esp8266 platform is not a priority ... while the esp32 has all the focus ...

sfranzyshen, 2019-11-11:
Using library elk at version 0.0.8 in folder: /home/name/Arduino/libraries/elk
Same result.

sfranzyshen, 2019-11-12:
Using library elk at version 0.0.9 in folder: /home/name/Arduino/libraries/elk
Same result.

sfranzyshen, 2019-11-14:
Using library elk at version 0.0.10 in folder: /home/name/Arduino/libraries/elk
section `.text1' will not fit in region `iram1_0_seg'

sfranzyshen, 2019-11-14:
`#define ESP_CONST_DATA __attribute__((aligned(4))) __attribute__((section(.irom.text)))`

sfranzyshen, 2019-11-14:
Using library elk at version 0.0.11 in folder: /home/name/Arduino/libraries/elk 
section .text1' will not fit in region iram1_0_seg'

cpq, 2019-11-25:
0.0.15 fixes an issue

closed 2019-11-25 by cpq

sfranzyshen, 2019-11-25:
LINUX_DIST="Linux Mint 18.3 Sylvia MATE 32-bit"
KERNEL=Linux computer 4.10.0-38-generic i686 i686 i686 GNU/Linux

ARDUINO IDE: 1.8.10 (32bit)
Elk Library: release 0.0.15 (Installed manually into Arduino IDE library folder)
ESP8266 BOARD: 2.5.2 (Installed with board manager from Arduino IDE)

Using library elk at version 0.0.15 in folder: /home/name/Arduino/libraries/elk 

Sketch uses 277964 bytes (26%) of program storage space. Maximum is 1044464 bytes.
Global variables use 28556 bytes (34%) of dynamic memory, leaving 53364 bytes for local variables. Maximum is 81920 bytes.

now compiles ... but crashes ...

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld

Exception (9):
epc1=0x40224814 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffef875 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd40 end: 3fffffc0 offset: 01a0
3ffffee0:  3ffef875 ff800000 00000003 3ffef878  
3ffffef0:  00000003 00000003 feefeffe feefeffe  
3fffff00:  feefeffe feefeffe feefeffe 3ffee9e0  
3fffff10:  40226af0 ffa00000 3ffef7ec 40100298  
3fffff20:  feefeffe 3ffef7ec 3ffef7ec 40226b8d  
3fffff30:  3ffe88be 00000000 3ffe88bd 402025be  
3fffff40:  40202921 00000001 3ffef6cc 3ffee9e0  
3fffff50:  3fffdad0 00000000 3ffee988 3ffee9e0  
3fffff60:  3fffdad0 00000000 3ffef7ec 402241c6  
3fffff70:  3ffe882a 3ffe882e 3ffee988 3ffee9e0  
3fffff80:  3fffdad0 3ffe882e 3ffef7ec 40226c23  
3fffff90:  feefeffe 00000000 3ffee988 402010c1  
3fffffa0:  3fffdad0 00000000 3ffee9b0 40201870  
3fffffb0:  feefeffe feefeffe 3ffe84f4 401006e5  
<<<stack<<<

```
```
Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
PC: 0x40224814
EXCVADDR: 0x3ffef875

Decoding stack results
0x402025be: uart_write(uart_t*, char const*, size_t) at /home/name/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/uart.cpp line 498
0x40202921: uart_init(int, int, int, int, int, size_t) at /home/name/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/uart.cpp line 661
0x402010c1: setup() at /home/name/Arduino/ArduinoBlink/ArduinoBlink.ino line 12
0x40201870: loop_wrapper() at /home/name/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/core_esp8266_main.cpp line 122

```

sfranzyshen, 2019-11-25:
after removing uart functions ... still getting 

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld

```

over and over ...