cesanta/elk #8: AVR under the Arduino IDE does not work

sfranzyshen, 2019-11-10:
LINUX_DIST="Linux Mint 18.3 Sylvia MATE 32-bit"
KERNEL=Linux computer 4.10.0-38-generic i686 i686 i686 GNU/Linux

ARDUINO IDE: 1.8.10 (32bit)
mDash Library: 1.1.8 (Installed using Arduino IDE library manager)
Arduino AVR boards (Nano): 1.8.1 (Installed with board manager from Arduino IDE)

**Test Code:**
```
  Serial.println(js_eval(js,
          "while (1) { "
          "  myWrite(13, 0); "
          "  myDelay(100); "
          "  myWrite(13, 1); "
          "  myDelay(100); "
          "}",
          -1));
```
**Responce:** This code should block and loop ... it doesn't it returns this right away ...
`4286578688`

sfranzyshen, 2019-11-11:
Using library elk at version 0.0.8 in folder: /home/name/Arduino/libraries/elk
Same result.

sfranzyshen, 2019-11-12:
Using library elk at version 0.0.9 in folder: /home/name/Arduino/libraries/elk
Same result.

cpq, 2019-11-14:
Released 0.0.10
Verified on Uno and Nano. Note - the .ino example has been updated, but README has not yet.
I had to copy src/avr into src/atmega328p to make it compile.

closed 2019-11-14 by cpq

sfranzyshen, 2019-11-14:
Using library elk at version 0.0.10 in folder: /home/name/Arduino/libraries/elk 

Sketch uses 28618 bytes (93%) of program storage space. Maximum is 30720 bytes.
Global variables use 727 bytes (35%) of dynamic memory, leaving 1321 bytes for local variables. Maximum is 2048 bytes.

confirmed working ... thank you!

sfranzyshen, 2019-11-25:
LINUX_DIST="Linux Mint 18.3 Sylvia MATE 32-bit"
KERNEL=Linux computer 4.10.0-38-generic i686 i686 i686 GNU/Linux

ARDUINO IDE: 1.8.10 (32bit)
elk Library: version 0.0.15 (Installed manually)
Arduino AVR boards (Nano): 1.8.1 (Installed with board manager from Arduino IDE)

Sketch uses 28286 bytes (92%) of program storage space. Maximum is 30720 bytes.
Global variables use 924 bytes (45%) of dynamic memory, leaving 1124 bytes for local variables. Maximum is 2048 bytes.

confirmed working ...