cesanta/elk #10: ESP32 under the Arduino IDE does not work ...

sfranzyshen, 2019-11-21:
LINUX_DIST="Linux Mint 18.3 Sylvia MATE 32-bit"
KERNEL=Linux computer 4.10.0-38-generic i686 i686 i686 GNU/Linux

ARDUINO IDE: 1.8.10 (32bit)
elk Library: version 0.0.11 (Installed using Arduino IDE library manager)
ESP32 boards (Nano): 1.0.4 (Installed with board manager from Arduino IDE)

**Test Code**
```
#include "elk.h"  // Add Elk library

extern "C" void myDelay(int milli) { delay(milli); }
extern "C" void myWrite(int pin, int val) { digitalWrite(pin, val); }
extern "C" void myMode(int pin, int mode) { pinMode(pin, mode); }

struct js *js;

void setup() {
  Serial.begin(115200);
  Serial.println();
  js = js_create(malloc(700), 700);
  Serial.println("malloc done");
  js_import(js, "f1", (uintptr_t) myDelay, "vi");
  js_import(js, "f2", (uintptr_t) myWrite, "vii");
  js_import(js, "f3", (uintptr_t) myMode, "vii");
  Serial.println("imports done");
  Serial.println(js_eval(js, "f3(2, 1);", 0));  // Set LED pin to OUTPUT mode
}

void loop() {
  Serial.println(js_eval(js, "f1(200); f2(2, 1); f1(200); f2(2, 0);", -1));
}
```

**Response**
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

malloc done
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d18e7  PS      : 0x00060130  A0      : 0x800d3461  A1      : 0x3ffb1f00  
A2      : 0x00000000  A3      : 0x3f401333  A4      : 0x00000002  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x69760031  A9      : 0x3ffb1f20  
A10     : 0x00000002  A11     : 0x3f4014eb  A12     : 0x0000000b  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000002c  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400d18e7:0x3ffb1f00 0x400d345e:0x3ffb1f20 0x400d0f09:0x3ffb1f80 0x400d3d4f:0x3ffb1fb0 0x40088215:0x3ffb1fd0

```

cpq, 2019-11-24:
Please try with 0.0.13.
Note - use `"f3(2, 2);` on ESP32, cause OUTPUT constant on ESP32 is 2, not 1 like on AVRs.

cpq, 2019-11-25:
Closing - please reopen if still fails.

closed 2019-11-25 by cpq

sfranzyshen, 2019-11-25:
confirmed to compile and run the test code correctly ... TY

```
Using library elk at version 0.0.15 in folder: /home/name/Arduino/libraries/elk 

Sketch uses 231185 bytes (17%) of program storage space. Maximum is 1310720 bytes.
Global variables use 16396 bytes (5%) of dynamic memory, leaving 311284 bytes for local variables. Maximum is 327680 bytes.

```