cesanta/elk #11: ESP8266 under the Arduino IDE does not work

sfranzyshen, 2019-11-25:
LINUX_DIST="Linux Mint 18.3 Sylvia MATE 32-bit"
KERNEL=Linux computer 4.10.0-38-generic i686 i686 i686 GNU/Linux

ARDUINO IDE: 1.8.10 (32bit)
Elk Library: release 0.0.15 (Installed manually into Arduino IDE library folder)
ESP8266 BOARD: 2.5.2 (Installed with board manager from Arduino IDE)

**Test Code**
```
#include "elk.h"  // Add Elk library

extern "C" void myDelay(int milli) { delay(milli); }
extern "C" void myWrite(int pin, int val) { digitalWrite(pin, val); }
extern "C" void myMode(int pin, int mode) { pinMode(pin, mode); }

struct js *js;

void setup() {
  js = js_create(malloc(700), 700);
  js_import(js, "f1", (uintptr_t) myDelay, "vi");
  js_import(js, "f2", (uintptr_t) myWrite, "vii");
  js_import(js, "f3", (uintptr_t) myMode, "vii");
  js_eval(js, "f3(2, 1);", 0);  // Set LED pin to OUTPUT mode ... tried f3(2, 2) also ...
}

void loop() {
  js_eval(js, "f1(200); f2(2, 1); f1(200); f2(2, 0);", 0);
}
```
just reboots ... 
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
```
over and over ...

cpq, 2019-11-25:
Hmm, smells like unaligned access exception. Elk uses unaligned pointers, maybe that makes ESP8266 sad. Looking for a way to deal with it..

sfranzyshen, 2019-11-25:
> Hmm, smells like unaligned access exception. Elk uses unaligned pointers, maybe that makes ESP8266 sad. Looking for a way to deal with it..

**compiler option**
`-mforce-l32
`
**attributes**
`__attribute__((aligned(4)))
`
???

cpq, 2019-11-27:
Released 0.0.16 which fixes many - maybe all, ESP8266 unaligned accesses.
Closing, please reopen if you see crashes.

closed 2019-11-27 by cpq

sfranzyshen, 2019-11-27:
I can confirm that 0.0.16 **builds** and **runs** on the **esp8266** platform ... TY!

`Using library elk at version 0.0.16 in folder: /home/name/Arduino/libraries/elk 
`
```
Sketch uses 280664 bytes (26%) of program storage space. Maximum is 1044464 bytes.
Global variables use 28552 bytes (34%) of dynamic memory, leaving 53368 bytes for local variables. Maximum is 81920 bytes.
```