cesanta/elk #13: 0.0.17 breaks nested while loops  on the esp32 platform ...

sfranzyshen, 2019-12-03:
I have code that was working correctly with 0.0.16 on the esp32 platform that stopped working with version 0.0.17 ... the problem seams to be with nested while loops ... this code was working on the esp32 ... untested on other platforms

```
js_eval(js, "let theaterChase = 
function(color, wait) { 
    a = 0; 
    while(a < 10) { 
        b = 0; 
        while(b < 3) { 
            clear(); 
            c = b; 
            while(c < numPixels()) { 
                setPixelColor32(c, color); c = c + 3; } 
                show(); 
                delay(wait); 
                b++; 
            } 
            a++; 
        } 
}; 
", 0);

```

cpq, 2019-12-07:
Nested loops seem to work fine:

```bash
$ cc -o elk examples/posix/main.c -I src -L src/macos -ldl -lelk
$ ./elk -e "let a=0, i=0; while(i++ < 3) { let j=0; while (j++ < 3) a++; } a"
9
```

In your code, I do not see `let` statements - are you referring to the global variables from within your function?

sfranzyshen, 2019-12-09:
yes, i am using global variables ... the in-function let statements didn't seam to work ... I'm going to look closer at this tonight and use the suggestion from the the other open issue to print out the results ...

sfranzyshen, 2019-12-10:
see #14 for update to this issue too!

sfranzyshen, 2019-12-18:
Your example above stops short of the problem ... the following does not work.

```
function(d) {
    let a = 0, b, c;
    while(a < 10) {
        b = 0; 
        while(b < 3) { 
            c = b; 
            while(c < 8) { 
                c += 3;
            } 
            b++;
        }
        a++;
    } 
    return a; 
};
```

cpq, 2019-12-18:
0.0.18 apparently fixes an issue.

closed 2019-12-18 by cpq