cesanta/elk #14: Bitwise AND operator

sfranzyshen, 2019-12-03:
LINUX_DIST="Linux Mint 18.3 Sylvia MATE 32-bit"
KERNEL=Linux computer 4.10.0-38-generic i686 i686 i686 GNU/Linux

ARDUINO IDE: 1.8.10 (32bit)
elk Library: version 0.0.16 (Installed using Arduino IDE library manager)
ESP32 boards (DEVKIT): 1.0.4 (Installed with board manager from Arduino IDE)

I'm having difficulty with some code I'm trying to get working on the esp32 ... I believe my problem stems from my use of the Bitwise AND operator ... here is a snippet 
**code snippet**
```
js_eval(js, "let rainbow =

function(wait) { 
    b = 0; 
    while(b < 1280) { 
        c = 0; 
        while(c < numPixels()) { 
            setPixelColor32(c, Wheel((c * 256 / numPixels()) + b));    
            c++; 
        } 
        show(); 
        delay(wait); 
        b++; 
    } 
}; 
", 0);
```

cpq, 2019-12-05:
Just to make sure:
elk integet is 23-bit, not 32 bit. Is numPixels() value falls into that expected range?
Could this test be shortened into a one-liner test?

sfranzyshen, 2019-12-06:
> Just to make sure:
> elk integet is 23-bit, not 32 bit. Is numPixels() value falls into that expected range?
> Could this test be shortened into a one-liner test?

in this case numPixels()  always returns 8 ... i believe that to be in range

cpq, 2019-12-07:
In you C code, capture the result of execution and print it - that could help to catch some obvious errors:

```c
  jsval_t v = js_eval(js, " this triggers error ", 0);", 0);
  printf("result: %s\n", js_str(js, v));
  js_gc(js, v);
```

sfranzyshen, 2019-12-09:
Yes I am referencing global variables because when I tried to use in-function let statements it didn't work ... the variables seamed to interfering with each other ... or at least it appeared that way ... perhaps like you suggested I can print the results out to get a better understanding of what's happening ... and I'll post full code used instead of snippets ... sometime tonight ... cheers and thanks!

cpq, 2019-12-10:
Thank you!
You can run quick experiments on linux or macos if you wish - see at the bottom of README on how to compile elk binary. You could also modify examples/posix/main.c in order to import custom C functions.

sfranzyshen, 2019-12-10:
I've tried to cover everything within this test code addressing both the open issues ...
I'm still having problems with nested variables (or nested while loops ... or both) and bitwise operators within 0.0.17 ... that I don't experience within 0.0.16 ... but I have problems with variables in both versions ... perhaps my approach is inaccurate?  

LINUX_DIST="Linux Mint 18.3 Sylvia MATE 32-bit"
KERNEL=Linux computer 4.10.0-38-generic i686 i686 i686 GNU/Linux

ARDUINO IDE: 1.8.10 (32bit)
elk Library: version 0.0.17 (Installed using Arduino IDE library manager)
ESP32 boards (DEVKIT): 1.0.4 (Installed with board manager from Arduino IDE)

**(Full)Test Code 0.0.17**
```
// example of using the elk javascript engine innen Arduino Environment on esp32
// Adafruit_NeoPixel strandtest.ino ported to elk javascript engine ...

#include <Adafruit_NeoPixel.h>
#include <elk.h> 

#define ALLOC     4096 // also tried 8192
#define LED_PIN   15
#define LED_COUNT 8
#define DEBUG     1

struct  js *js = js_create(malloc(ALLOC), ALLOC);

// NeoPixel config could be handled elsewhere ... such as with a web interface

Adafruit_NeoPixel strip(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

extern "C" void Sys_delay(unsigned long milli) {
  delay(milli);
}

extern "C" void Sys_print(const char *str) {
  Serial.println(str);
}

extern "C" uint32_t Neo_getPixelColor(uint16_t n) {
  return strip.getPixelColor(n);
}

extern "C" uint32_t Neo_Color(uint8_t r, uint8_t g, uint8_t b) {
  return strip.Color(r, g, b);
}

extern "C" void Neo_setPixelColor(uint16_t n, uint32_t c) {
  strip.setPixelColor(n, c);
}

extern "C" uint16_t Neo_numPixels(void) {
  return strip.numPixels();
}

extern "C" uint8_t Neo_getBrightness(void) {
  return strip.getBrightness();
}

extern "C" void Neo_setBrightness(uint8_t brightness) {
  strip.setBrightness(brightness);
}

extern "C" void Neo_fill(uint32_t c=0, uint16_t first=0, uint16_t count=0) {
  strip.fill(c=0, first=0, count=0);
}

extern "C" void Neo_show() {
  strip.show();
}

extern "C" void Neo_clear() {
[... 759 lines cut ...]
    while(j<256) {
        i=0;
        while(i<strip.numPixels()) {
            strip.setPixelColor(i, Wheel((i+j) & 255));
            i++;
        }
        strip.show();
        delay(wait);
        j++;
    }
}

// c for
void rainbow(uint8_t wait) {
  uint16_t i, j;

  for(j=0; j<256; j++) {
    for(i=0; i<strip.numPixels(); i++) {
      strip.setPixelColor(i, Wheel((i+j) & 255));
    }
    strip.show();
    delay(wait);
  }
}
```

sfranzyshen, 2019-12-11:
When I edit the while loops to have the let statements within them rather than at the top of each function ... for example:

from this ...
`  v = js_eval(js, "let theaterChase = function(color, wait) { let a = 0, b, c; while(a < 10) { b = 0; while(b < 3) { clear(); c = b; while(c < numPixels()) { setPixelColor(c, color); c = c + 3; } show(); delay(wait); b++; } a++; } return a; };", 0);
`

to this ...
`  v = js_eval(js, "let theaterChase = function(color, wait) { let a = 0; while(a < 10) { let b = 0; while(b < 3) { clear(); let c = b; while(c < numPixels()) { setPixelColor(c, color); c = c + 3; } show(); delay(wait); b++; } a++; } return c; };", 0);
`

I get this response ...
in 0.0.16
```
Starting-up ...
let colorWipe
result: function(color, wait) { let i = 0; while(i < numPixels()) { setPixelColor(i, color); show(); delay(wait); i++; } return i; }
let theaterChase
result: function(color, wait) { let a = 0; while(a < 10) { let b = 0; while(b < 3) { clear(); let c = b; while(c < numPixels()) { setPixelColor(c, color); c = c + 3; } show(); delay(wait); b++; } a++; } return c; }
let theaterChaseRainbow
result: function(wait) { let j = 0; while(j < 256) { let q = 0; while(q < 3) { let i = 0; while(i < numPixels()) { setPixelColor(i+q, Wheel( (i+j) % 255)); i = i + 3; } show(); delay(wait); i = 0; while(i < numPixels()) { setPixelColor(i+q, 0); i = i + 3; } q++; 
let rainbowCycle
result: function(wait) { let j = 0; while(j < 256 * 5) { let i = 0; while(i < numPixels()) { setPixelColor(i, Wheel(((i * 256 / numPixels()) + j) & 255)); i++; } show(); delay(wait); j++; } return i; }
let rainbow
result: function(wait) { let j = 0; while(j < 256) { let i = 0; while(i < numPixels()) { setPixelColor(i, Wheel((i + j) & 255)); i++; } show(); delay(wait); j++; } return i; }
let loop
result: function() { colorWipe(Color(255, 0, 0), 50); colorWipe(Color(0, 255, 0), 50); colorWipe(Color(0, 0, 255), 50); colorWipe(Color(255, 255, 255), 50); theaterChase(Color(255, 255, 255), 50); theaterChase(Color(255, 0, 0), 50); theaterChase(Color(0, 0, 255),
jsinfo()
{"core":136,"mem":3960,"used":2011}
result: "Done"
setup complete ... looping
 ... looping
loop()
result: ERROR 1145
jsinfo()
{"core":136,"mem":3960,"used":2060}
result: "Done"
colorWipe()
result: "Done"
theaterChase()
result: ERROR 1145
rainbow()
result: 0
rainbowCycle()
result: ERROR 1278
theaterChaseRainbow()
result: undefined
 ... looping
loop()
result: ERROR 1190
jsinfo()
{"core":136,"mem":3960,"used":2211}
result: "Done"
colorWipe()
result: "Done"
theaterChase()
result: "Done"
rainbow()
result: ERROR 1190
rainbowCycle()
result: ERROR 1190
theaterChaseRainbow()
result: undefined
 ... looping
loop()
result: ERROR 1190
jsinfo()
{"core":136,"mem":3960,"used":2211}
result: "Done"
colorWipe()
result: "Done"
theaterChase()
result: "Done"
rainbow()
result: ERROR 1190
rainbowCycle()
result: ERROR 1190
theaterChaseRainbow()
result: undefined
 ... looping
loop()
result: ERROR 1190
jsinfo()
{"core":136,"mem":3960,"used":2211}
result: "Done"
colorWipe()
result: "Done"
theaterChase()
result: "Done"
rainbow()
result: ERROR 1190
rainbowCycle()
result: ERROR 1190
theaterChaseRainbow()
result: undefined
 ... looping

```
and this in 0.0.17
```
Starting-up ...
let colorWipe
result: function(color, wait) { let i = 0; while(i < numPixels()) { setPixelColor(i, color); show(); delay(wait); i++; } return i; }
let theaterChase
result: function(color, wait) { let a = 0; while(a < 10) { let b = 0; while(b < 3) { clear(); let c = b; while(c < numPixels()) { setPixelColor(c, color); c = c + 3; } show(); delay(wait); b++; } a++; } return c; }
let theaterChaseRainbow
result: function(wait) { let j = 0; while(j < 256) { let q = 0; while(q < 3) { let i = 0; while(i < numPixels()) { setPixelColor(i+q, Wheel( (i+j) % 255)); i = i + 3; } show(); delay(wait); i = 0; while(i < numPixels()) { setPixelColor(i+q, 0); i = i + 3; } q++; } j++; } return i; }
let rainbowCycle
result: function(wait) { let j = 0; while(j < 256 * 5) { let i = 0; while(i < numPixels()) { setPixelColor(i, Wheel(((i * 256 / numPixels()) + j) & 255)); i++; } show(); delay(wait); j++; } return i; }
let rainbow
result: function(wait) { let j = 0; while(j < 256) { let i = 0; while(i < numPixels()) { setPixelColor(i, Wheel((i + j) & 255)); i++; } show(); delay(wait); j++; } return i; }
let loop
result: function() { colorWipe(Color(255, 0, 0), 50); colorWipe(Color(0, 255, 0), 50); colorWipe(Color(0, 0, 255), 50); colorWipe(Color(255, 255, 255), 50); theaterChase(Color(255, 255, 255), 50); theaterChase(Color(255, 0, 0), 50); theaterChase(Color(0, 0, 255), 50); theaterChase(Color(0, 255, 0), 50); rainbow(10); rainbowCycle(10); theaterChaseRainbow(50); return 'Done'; }
jsinfo()
{"core":136,"mem":3960,"used":1861}
result: "Done"
setup complete ... looping
 ... looping
loop()
result: ERROR 1195
jsinfo()
{"core":136,"mem":3960,"used":2047}
result: "Done"
colorWipe()
result: "Done"
theaterChase()
result: "Done"
rainbow()
result: ERROR 1195
rainbowCycle()
result: ERROR 1195
theaterChaseRainbow()
result: undefined
 ... looping
loop()
result: ERROR 1195
jsinfo()
{"core":136,"mem":3960,"used":2047}
result: "Done"
colorWipe()
result: "Done"
theaterChase()
result: "Done"
rainbow()
result: ERROR 1195
rainbowCycle()
result: ERROR 1195
theaterChaseRainbow()
result: undefined
 ... looping
loop()
result: ERROR 1195
jsinfo()
{"core":136,"mem":3960,"used":2047}
result: "Done"
colorWipe()
result: "Done"
theaterChase()
result: "Done"
rainbow()
result: ERROR 1195
rainbowCycle()
result: ERROR 1195
theaterChaseRainbow()
result: undefined
 ... looping

```

cpq, 2019-12-17:
Are you sure you're running the latest version?
The latest version does not export js_fmt.

sfranzyshen, 2019-12-17:
> Are you sure you're running the latest version?
> The latest version does not export js_fmt.

I performed **all** test on both 0.0.16 & 0.0.17 ... shown above (0.0.17 code is listed first) is the code used for the 0.0.17 test ... and (second) the next block of code was used for testing 0.0.16 ... so ... **Yes**, I'm running the latest version! ... **and** the 0.0.16 older version too! ... I was trying to show how somethings were working under 0.0.16 ... and are now not working in 0.0.17 ... 

sorry for all the content above ... and if it caused any confusion (I tried to label everything ...)

cpq, 2019-12-17:
Ah, sorry about that.

Also please note that elk's numbers are packed into the C 32-bit `float` type. Thus, integer values are 23 bit max. You cannot import functions that return or accept `uint32_t`. Make the range fall into 23 bits. For example, instead of `setPixelColor()` that accept a single `uint32_t` color value, make it accept 3 values - r, g, b; each taking 0-255 range.

Of course this point does not address an issue you've reported. We'll try to reproduce.

cpq, 2019-12-17:
Starting to repro. This works fine on my 16-pixel strip:

```c++
// example of using the elk javascript engine innen Arduino Environment on esp32
// Adafruit_NeoPixel strandtest.ino ported to elk javascript engine ...

#include <Adafruit_NeoPixel.h>
#include <elk.h> 

#define ALLOC     4096 // also tried 8192
#define LED_PIN   10
#define LED_COUNT 16
#define DEBUG     1

struct  js *js = js_create(malloc(ALLOC), ALLOC);
Adafruit_NeoPixel strip(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println("Starting-up ...");

  js_import(js, "delay", (uintptr_t) Sys_delay, "vi");
  js_import(js, "print", (uintptr_t) Sys_print, "vs");
  js_import(js, "show", (uintptr_t) Neo_show, "v");
  js_import(js, "setPixelColor", (uintptr_t) Neo_setPixelColor, "viiii");

  js_eval(js, "let colorWipe = function(d, r, g, b) { let i = 0; while (i < 16) { i++; setPixelColor(i, r, g, b); show(); delay(d); } };", 0);

  strip.begin();
  strip.show();
  strip.setBrightness(50);
  Serial.println("setup complete ...");
}

extern "C" void Neo_show() {  strip.show(); }
extern "C" void Neo_setPixelColor(uint16_t n, int r, int g, int b) { strip.setPixelColor(n, strip.Color(r, g, b)); }
extern "C" void Sys_delay(unsigned long milli) { delay(milli); }
extern "C" void Sys_print(const char *str) { Serial.println(str); }

void loop() {
  Serial.println(" ... looping");
  js_eval(js, "colorWipe(50, 255, 0, 0);", 0);
  js_eval(js, "colorWipe(50, 0, 255, 0);", 0);
}
```

cpq, 2019-12-17:
Added theaterChase() which also works fine (see below).
Please refactor the rest of the tests in the same manner. Please let me know how that works.

```c++
// example of using the elk javascript engine innen Arduino Environment on esp32
// Adafruit_NeoPixel strandtest.ino ported to elk javascript engine ...

#include <Adafruit_NeoPixel.h>
#include <elk.h> 

#define ALLOC     4096 // also tried 8192
#define LED_PIN   10
#define LED_COUNT 16
#define DEBUG     1

struct  js *js = js_create(malloc(ALLOC), ALLOC);
Adafruit_NeoPixel strip(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println("Starting-up ...");

  js_import(js, "delay", (uintptr_t) Sys_delay, "vi");
  js_import(js, "print", (uintptr_t) Sys_print, "vs");
  js_import(js, "show", (uintptr_t) Neo_show, "v");
  js_import(js, "clear", (uintptr_t) Neo_clear, "v");
  js_import(js, "setPixelColor", (uintptr_t) Neo_setPixelColor, "viiii");

  js_eval(js, "let colorWipe = function(d, r, g, b) { let i = 0; while (i < 16) { i++; setPixelColor(i, r, g, b); show(); delay(d); } };", 0);
  js_eval(js, "let theaterChase = function(d, r, g, b) { let a = 0, b, c; while (a < 10) { b = 0; while (b < 3) { clear(); c=b; while (c<16) { setPixelColor(c, r,g,b); c += 3; } show(); delay(d); b++; } a++; } };", 0);
  
  strip.begin();
  strip.show();
  strip.setBrightness(50);
  Serial.println("setup complete ...");
}

extern "C" void Neo_show() {  strip.show(); }
extern "C" void Neo_clear() {  strip.clear(); }
extern "C" void Neo_setPixelColor(uint16_t n, int r, int g, int b) { strip.setPixelColor(n, strip.Color(r, g, b)); }
extern "C" void Sys_delay(unsigned long milli) { delay(milli); }
extern "C" void Sys_print(const char *str) { Serial.println(str); }

void loop() {
  Serial.println(" ... looping");
  js_eval(js, "colorWipe(50, 255, 0, 0);", 0);
  js_eval(js, "colorWipe(50, 0, 255, 0);", 0);
  js_eval(js, "theaterChase(50, 255, 0, 0);", 0);
}
```

sfranzyshen, 2019-12-18:
> Added theaterChase() which also works fine (see below).
> Please refactor the rest of the tests in the same manner. Please let me know how that works.
>
> ```c++
> // example of using the elk javascript [...]

The ColorWipe works fine every time ... however ... i'm clearly still getting different results for the theaterChase ... the first three times loop() is called theaterChase does not finish the while loop correctly (it's ending early) then i get strange colors on the first and second leds ... that I don't experience run a straight c version of the code ... i'll print some things out to see what's going on ...

sfranzyshen, 2019-12-18:
LINUX_DIST="Linux Mint 18.3 Sylvia MATE 32-bit"
KERNEL=Linux computer 4.10.0-38-generic i686 i686 i686 GNU/Linux

ARDUINO IDE: 1.8.10 (32bit)
elk Library: version 0.0.16 (Installed using Arduino IDE library manager)
ESP32 boards (DEVKIT): 1.0.4 (Installed with board manager from Arduino IDE)

Here I have (isolated) theaterChase() echoing back to the console ... and you can clearly see it's not working as expected ... the function never (returns) reaches the end of all the loops ... it bombs out after a couple outer and inner loops ... see below for the results

**TEST CODE**
```
// example of using the elk javascript engine innen Arduino Environment on esp32
// Adafruit_NeoPixel strandtest.ino ported to elk javascript engine ...

#include <Adafruit_NeoPixel.h>
#include <elk.h> 

#define ALLOC     4096 // also tried 8192
#define LED_PIN   18
#define LED_COUNT 8
#define DEBUG     1

struct  js *js = js_create(malloc(ALLOC), ALLOC);
Adafruit_NeoPixel strip(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  jsval_t v;

  Serial.begin(115200);
  Serial.println();
  Serial.println("Starting-up ...");

  js_import(js, "delay", (uintptr_t) Sys_delay, "vi");
  js_import(js, "printStr", (uintptr_t) Sys_printStr, "vs");
  js_import(js, "printInt", (uintptr_t) Sys_printInt, "vi");
  js_import(js, "show", (uintptr_t) Neo_show, "v");
  js_import(js, "clear", (uintptr_t) Neo_clear, "v");
  js_import(js, "setPixelColor", (uintptr_t) Neo_setPixelColor, "viiii");
  js_import(js, "jsinfo", (uintptr_t) js_info, "sm");

  v = js_eval(js, "let theaterChase = function(d, rd, gr, bl) { let a = 0, b, c; while(a < 10) { printStr('a = '); printInt(a); b = 0; while(b < 3) { printStr('b = '); printInt(b); clear(); c = b; while(c < 8) { printStr('c = '); printInt(c); setPixelColor(c, rd, gr, bl); c += 3; } show(); delay(d); b++; } a++; } return 'Done'; };", 0);

  if(DEBUG) {
    Serial.print("let theaterChase ");
    Serial.printf("result: %s\n", js_str(js, v));
  }  
  
  js_gc(js, v);
  
  strip.begin();
  strip.show();
  strip.setBrightness(50);

  v = js_eval(js, "printStr(jsinfo(null)); return 'Done';", 0);

  if(DEBUG) {
    Serial.println("jsinfo()");
    Serial.printf("result: %s\n", js_str(js, v));
  }
  
  js_gc(js, v);

  Serial.println("setup complete ...");
}

extern "C" void Neo_show() { Serial.println("show :"); strip.show(); }
extern "C" void Neo_clear() { Serial.println("clear :"); strip.clear(); }
extern "C" void Neo_setPixelColor(uint16_t n, int r, int g, int b) { Serial.print("setPixelColor :"); Serial.println(n); strip.setPixelColor(n, strip.Color(r, g, b)); }
extern "C" void Sys_delay(unsigned long milli) { Serial.print("delay :"); Serial.println(milli); delay(milli); }
extern "C" void Sys_printStr(const char *str) { Serial.println(str); }
extern "C" void Sys_printInt(int i) { Serial.println(i); }

void loop() {
  jsval_t v;

  Serial.println(" ... looping");

  v = js_eval(js, "theaterChase(50, 255, 0, 0);", 0);

  if(DEBUG) {
    Serial.println("theaterChase() ");
    Serial.printf("result: %s\n", js_str(js, v));
  }
  js_gc(js, v);

}
```
**RESULT**

```
Starting-up ...
let theaterChase result: function(d, rd, gr, bl) { let a = 0, b, c; while(a < 10) { printStr('a = '); printInt(a); b = 0; while(b < 3) { printStr('b = '); printInt(b); clear(); c = b; while(c < 8) { printStr('c = '); printInt(c); setPixelColor(c, rd, gr, bl); c += 3; } show(); delay(d); b++; } a++; } return 'Done'; }
{"core":136,"mem":3960,"used":539}
jsinfo()
result: "Done"
setup complete ...
 ... looping
a = 
0
b = 
0
clear :
c = 
0
setPixelColor :0
c = 
3
setPixelColor :3
c = 
6
setPixelColor :6
show :
delay :50
b = 
1
clear :
c = 
1
setPixelColor :1
c = 
4
setPixelColor :4
c = 
7
setPixelColor :7
show :
delay :50
b = 
2
clear :
c = 
2
setPixelColor :2
c = 
5
setPixelColor :5
show :
delay :50
a = 
1
b = 
0
clear :
c = 
0
setPixelColor :0
c = 
3
setPixelColor :3
c = 
6
setPixelColor :6
show :
delay :50
b = 
1
clear :
c = 
1
setPixelColor :1
c = 
theaterChase() 
result: undefined
 ... looping
a = 
0
b = 
0
clear :
c = 
0
setPixelColor :0
c = 
3
setPixelColor :3
c = 
6
setPixelColor :6
show :
delay :50
b = 
1
clear :
c = 
1
setPixelColor :1
c = 
4
setPixelColor :4
c = 
7
setPixelColor :7
show :
delay :50
b = 
2
clear :
c = 
2
setPixelColor :2
c = 
5
setPixelColor :5
show :
delay :50
a = 
1
b = 
0
clear :
c = 
0
setPixelColor :0
c = 
3
setPixelColor :3
c = 
6
setPixelColor :6
show :
delay :50
b = 
1
clear :
c = 
1
setPixelColor :1
c = 
theaterChase() 
result: undefined
 ... looping
a = 
0
b = 
0
clear :
c = 
0
setPixelColor :0
c = 
3
setPixelColor :3
c = 
6
setPixelColor :6
show :
delay :50
b = 
1
clear :
c = 
1
setPixelColor :1
c = 
4
setPixelColor :4
c = 
7
setPixelColor :7
show :
delay :50
b = 
2
clear :
c = 
2
setPixelColor :2
c = 
5
setPixelColor :5
show :
delay :50
a = 
1
b = 
0
clear :
c = 
0
setPixelColor :0
c = 
3
setPixelColor :3
c = 
6
setPixelColor :6
show :
delay :50
b = 
1
clear :
c = 
1
setPixelColor :1
c = 
theaterChase() 
result: undefined
 ... looping

```

sfranzyshen, 2019-12-18:
stripped down to nothing but elk ... **clearly** still having issues with either variables or while loops or both

**TEST CODE**
```
#include <elk.h> 

#define ALLOC     4096 // also tried 8192
#define DEBUG     1

struct  js *js = js_create(malloc(ALLOC), ALLOC);

void setup() {
  jsval_t v;

  Serial.begin(115200);
  Serial.println();
  Serial.println("Starting-up ...");

  js_import(js, "delay", (uintptr_t) Sys_delay, "vi");
  js_import(js, "printStr", (uintptr_t) Sys_printStr, "vs");
  js_import(js, "printInt", (uintptr_t) Sys_printInt, "vi");
  js_import(js, "jsinfo", (uintptr_t) js_info, "sm");

  v = js_eval(js, "let theaterChase = function(d) { let a = 0, b, c; while(a < 10) { printStr('a = '); printInt(a); b = 0; while(b < 3) { printStr('b = '); printInt(b); c = b; while(c < 8) { printStr('c = '); printInt(c); c += 3; } b++; } a++; } return 'Done'; };", 0);

  if(DEBUG) {
    Serial.print("let theaterChase ");
    Serial.printf("result: %s\n", js_str(js, v));
  }  
  
  js_gc(js, v);
  
  v = js_eval(js, "printStr(jsinfo(null)); return 'Done';", 0);

  if(DEBUG) {
    Serial.println("jsinfo()");
    Serial.printf("result: %s\n", js_str(js, v));
  }
  
  js_gc(js, v);

  Serial.println("setup complete ...");
}

extern "C" void Sys_delay(unsigned long milli) { Serial.print("delay :"); Serial.println(milli); delay(milli); }
extern "C" void Sys_printStr(const char *str) { Serial.println(str); }
extern "C" void Sys_printInt(int i) { Serial.println(i); }

void loop() {
  jsval_t v;

  Serial.println(" ... looping");

  v = js_eval(js, "theaterChase(50);", 0);

  if(DEBUG) {
    Serial.println("theaterChase() ");
    Serial.printf("result: %s\n", js_str(js, v));
  }
  js_gc(js, v);

}
```

**RESULT**
```
Starting-up ...
let theaterChase result: function(d) { let a = 0, b, c; while(a < 10) { printStr('a = '); printInt(a); b = 0; while(b < 3) { printStr('b = '); printInt(b); c = b; while(c < 8) { printStr('c = '); printInt(c); c += 3; } b++; } a++; } return 'Done'; }
{"core":136,"mem":3960,"used":372}
jsinfo()
result: "Done"
setup complete ...
 ... looping
a = 
0
b = 
0
c = 
0
c = 
3
c = 
6
b = 
1
c = 
1
c = 
4
c = 
7
b = 
2
c = 
2
c = 
5
a = 
1
b = 
0
c = 
0
c = 
3
c = 
6
b = 
1
c = 
1
c = 
theaterChase() 
result: undefined
 ... looping

```

cpq, 2019-12-18:
There was an issue indeed, around loops scope handling. Released 0.0.18.
Now, the following sketch works on both ESP8266 and Uno:

```c++
#include "elk.h"

char mem[600];
struct  js *js = js_create(mem, sizeof(mem));
extern "C" void Sys_print(jsval_t v) { Serial.println(js_str(js, v)); }

void setup() {
  Serial.begin(115200);
  Serial.println("Starting-up ...");
  js_import(js, "print", (uintptr_t) Sys_print, "vj");

  js_eval(js, "let f = function(d) { let a = 0, b, c; while(a < 10) { print({a:a}); "
  "b = 0; while(b < 3) { print({b:b}); c = b; while(c < 8) { print({c:c}); c += 3; } b++; } a++; } };", 0);
}

void loop() {
  js_eval(js, "f();", 0);
  Serial.println(js_info(js));
  delay(1000);
}
```

sfranzyshen, 2019-12-18:
TY confirmed to be working in 0.0.18 :) @cpq

sfranzyshen, 2019-12-18:
LINUX_DIST="Linux Mint 18.3 Sylvia MATE 32-bit"
KERNEL=Linux computer 4.10.0-38-generic i686 i686 i686 GNU/Linux

ARDUINO IDE: 1.8.10 (32bit)
elk Library: version 0.0.18 (Installed using Arduino IDE library manager)
ESP32 boards (DEVKIT): 1.0.4 (Installed with board manager from Arduino IDE)

I still seam to be having a problem with bitwise operations throwing off execution order ... below is a snippet of the code I'll been working on to demonstrate what happens ... In the first example setPixelColor() gets called after WheelR() **BUT** before WheelG or WheelB are called ... so the g & b variables are always set to zero (0) ... setting it to a variable first solved the problem for me ... but it seams to still be a problem in some cases ...

**NOT WORKING**
```
v = js_eval(js, "let rainbow = 

function(d) { 
    let j = 0, i; 
    while(j < 256) { 
        i = 0; 
        while(i < numPixels()) { 
            setPixelColor(i, WheelR((i + j) & 255), WheelG((i + j) & 255), WheelB((i + j) & 255));
            i++;
        }
        show();
        delay(d);
        j++;
    }
    return j;
};

", 0);
```

**WORKING**
```
v = js_eval(js, "let rainbow = 

function(d) { 
    let j = 0, i, p;
    while(j < 256) {
        i = 0;
        while(i < numPixels()) { 
            p = (i + j) & 255; 
            setPixelColor(i, WheelR(p), WheelG(p), WheelB(p));
            i++;
        }
        show();
        delay(d);
        j++;
    }
    return j;
};

", 0);

```

sfranzyshen, 2019-12-19:
UPDATE: it now appears as if the problem isn't with the bitwise operation itself but rather with the function parameters and how they are being processed  (order) ... here is another example ... again in the first example the setPixelColor() function gets called before the WheelX() functions get called ... again leaving rgb (them) all set to zero ... not the desired outcome ... however when I pass the first parameter to setPixelColor() as a pre-calculated variable ... it functions as expected ...

**NOT WORKING**
```
v = js_eval(js, "let theaterChaseRainbow = 
function(wait) {
    let j = 0, q, i, p; 
    while(j < 256) {
        q = 0; 
        while(q < 3) {
            i = 0; 
            while(i < 8) {
                p = (i + j) % 255;
                setPixelColor(i + q, WheelR(p), WheelG(p), WheelB(p)); 
                i = i + 3;
            }
            show();
            delay(wait);
            i = 0;
            while(i < 8) {
                setPixelColor(i + q, 0, 0, 0);
                i = i + 3;
            }
            q++;
        }
        j++;
    }
    return j;
};

", 0);
```

**WORKING**
```
v = js_eval(js, "let theaterChaseRainbow = 
function(wait) {
    let j = 0, q, i, p, r; 
    while(j < 256) {
        q = 0; 
        while(q < 3) {
            i = 0; 
            while(i < 8) {
                p = (i + j) % 255;
                r = i + q;
                setPixelColor(r, WheelR(p), WheelG(p), WheelB(p)); 
                i = i + 3;
            }
            show();
            delay(wait);
            i = 0;
            while(i < 8) {
                r = i + q;
                setPixelColor(r, 0, 0, 0);
                i = i + 3;
            }
            q++;
        }
        j++;
    }
    return j;
};

", 0);
```

sfranzyshen, 2019-12-19:
Just for fun here's my final working code ...
```
// example of using the elk javascript engine innen Arduino Environment on esp32
// Adafruit_NeoPixel strandtest.ino ported to elk javascript engine ...

#include <Adafruit_NeoPixel.h>
#include <elk.h> 

#define ALLOC     4096 // also tried 8192
#define LED_PIN   18
#define LED_COUNT 8
#define DEBUG     0

struct  js *js = js_create(malloc(ALLOC), ALLOC);
Adafruit_NeoPixel strip(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  jsval_t v;

  Serial.begin(115200);
  Serial.println();
  
  if(DEBUG) {
    Serial.println("Starting-up ...");
  }

  js_import(js, "delay", (uintptr_t) Sys_delay, "vi");
  js_import(js, "printStr", (uintptr_t) Sys_printStr, "vs");
  js_import(js, "printInt", (uintptr_t) Sys_printInt, "vi");
  js_import(js, "show", (uintptr_t) Neo_show, "v");
  js_import(js, "clear", (uintptr_t) Neo_clear, "v");
  js_import(js, "setPixelColor", (uintptr_t) Neo_setPixelColor, "viiii");
  js_import(js, "WheelR", (uintptr_t) Neo_WheelR, "ii");
  js_import(js, "WheelG", (uintptr_t) Neo_WheelG, "ii");
  js_import(js, "WheelB", (uintptr_t) Neo_WheelB, "ii");
  js_import(js, "jsinfo", (uintptr_t) js_info, "sm");
  js_import(js, "numPixels", (uintptr_t) Neo_numPixels, "i");

  v = js_eval(js, "let colorWipe = function(d, r, g, b) { let i = 0; while (i < numPixels()) { setPixelColor(i, r, g, b); show(); delay(d); i++; } delay(d * numPixels()); return i; };", 0);
  if(DEBUG) {
    Serial.printf("let colorWipe result: %s\n", js_str(js, v));
  }
  js_gc(js, v);

  v = js_eval(js, "let theaterChase = function(d, r, g, bl) { let a = 0, b, c; while(a < 10) { b = 0; while(b < 3) { clear(); c = b; while(c < numPixels()) { setPixelColor(c, r, g, bl); c += 3; } show(); delay(d); b++; } a++; } return a; };", 0);
  if(DEBUG) {
    Serial.printf("let theaterChase result: %s\n", js_str(js, v));
  }
  js_gc(js, v);
 
  v = js_eval(js, "let theaterChaseRainbow = function(wait) { let j = 0, q, i, p, r; while(j < 256) { q = 0; while(q < 3) { i = 0; while(i < numPixels()) { p = (i + j) % 255; r = i + q; setPixelColor(r, WheelR(p), WheelG(p), WheelB(p)); i = i + 3; } show(); delay(wait); i = 0; while(i < numPixels()) { r = i + q; setPixelColor(r, 0, 0, 0); i = i + 3; } q++; } j++; } return j; }; ", 0);
  if(DEBUG) {
    Serial.printf("let theaterChaseRainbow result: %s\n", js_str(js, v));
  }
  js_gc(js, v);

  v = js_eval(js, "let rainbowCycle = function(wait) { let j = 0, i, p; while(j < 256 * 5) { i = 0; while(i < numPixels()) { p = ((i * 256 / numPixels()) + j) & 255; setPixelColor(i, WheelR(p), WheelG(p), WheelB(p)); i++; } show(); delay(wait); j++; } return j; };", 0);
  if(DEBUG) {
    Serial.printf("let rainbowCycle result: %s\n", js_str(js, v));
  }
  js_gc(js, v);
  
  v = js_eval(js, "let rainbow = function(d) { let j = 0, i, p; while(j < 256) { i = 0; while(i < numPixels()) { p = (i + j) & 255; setPixelColor(i, WheelR(p), WheelG(p), WheelB(p)); i++; } show(); delay(d); j++; } return j; };", 0);
  if(DEBUG) {
    Serial.printf("let rainbow result: %s\n", js_str(js, v));
  }
  js_gc(js, v);
   
  strip.begin();
  strip.show();
[... 96 lines cut ...]
  if(DEBUG) {
    Serial.printf("theaterChase() blue result: %s\n", js_str(js, v));
  }
  js_gc(js, v);
    
  v = js_eval(js, "rainbow(10);", 0);
  if(DEBUG) {
    Serial.printf("rainbow() result: %s\n", js_str(js, v));
  }
  js_gc(js, v);

  v = js_eval(js, "rainbowCycle(10);", 0);  
  if(DEBUG) {
    Serial.printf("rainbowCycle() result: %s\n", js_str(js, v));
  }
  js_gc(js, v);

  v = js_eval(js, "theaterChaseRainbow(50);", 0);
  if(DEBUG) {
    Serial.printf("theaterChaseRainbow() result: %s\n", js_str(js, v));
  }
  js_gc(js, v);
  
}
```

cpq, 2019-12-19:
Nice!
Nitpick: remove return outside of a function `v = js_eval(js, "printStr(jsinfo(null)); return 'Done';", 0);`.
Nitpick: use more compact `extern "C" void Sys_print(jsval_t v) { Serial.println(js_str(js, v)); }` for printing values.

Could that code work on Uno?

Since that is a different issue, please open a new one - I'd appreciate a minimal example.

sfranzyshen, 2019-12-19:
> Nice!
> Nitpick: remove return outside of a function `v = js_eval(js, "printStr(jsinfo(null)); return 'Done';", 0);`.
> Nitpick: use more compact `extern "C" void Sys_print(jsval_t v) { Serial.println(js_str(js, v)); }` for printing values.
> 
> Can that neopixel code work on Uno?
> 
> Since that is a different issue, please open a new one - I'd appreciate a minimal example.

I haven't tried this on anything else (yet!) but on the esp32 ... I'll try and get to it working today ...
I also opened issue #15 to continue with the "out of order" execution issues that persist ... I'll open another one with a "minimal example" that hopefully will work on **ALL** supported platforms 

Cheers and Thanks for the great work!

closed 2019-12-19 by sfranzyshen

cpq, 2019-12-21:
Could you reduce that to a one-liner please?