cesanta/elk #15: calculated function parameters are being processed out of order

sfranzyshen, 2019-12-19:
LINUX_DIST="Linux Mint 18.3 Sylvia MATE 32-bit"
KERNEL=Linux computer 4.10.0-38-generic i686 i686 i686 GNU/Linux

ARDUINO IDE: 1.8.10 (32bit)
elk Library: version 0.0.18 (Installed using Arduino IDE library manager)
ESP32 boards (DEVKIT): 1.0.4 (Installed with board manager from Arduino IDE)

This is a continuation of issue #14 ...

below are snippets of the code to demonstrate what happens ... In the first example setPixelColor() gets called after WheelR() **BUT** before WheelG() or WheelB() are called ... so the g & b variables are always set to zero (0) ... setting it to a variable first solved the problem ... but it may still be a problem in some cases ... 

**NOT WORKING**
```
v = js_eval(js, "let rainbow = 

function(d) { 
    let j = 0, i; 
    while(j < 256) { 
        i = 0; 
        while(i < numPixels()) { 
            setPixelColor(i, WheelR((i + j) & 255), WheelG((i + j) & 255), WheelB((i + j) & 255));
            i++;
        }
        show();
        delay(d);
        j++;
    }
    return j;
};

", 0);
```

**WORKING**
```
v = js_eval(js, "let rainbow = 

function(d) { 
    let j = 0, i, p;
    while(j < 256) {
        i = 0;
        while(i < numPixels()) { 
            p = (i + j) & 255; 
            setPixelColor(i, WheelR(p), WheelG(p), WheelB(p));
            i++;
        }
        show();
        delay(d);
        j++;
    }
    return j;
};

", 0);

```
Here the problem is also with the function parameters and how they are being processed  (order) ...  again in the first example the setPixelColor() function gets called before the any of the WheelX() functions get called ... again leaving rgb variables all set to zero ... this is not the desired outcome ... however when I pass the first parameter to setPixelColor() as a pre-calculated variable ... it functions as expected ...

**NOT WORKING**
```
v = js_eval(js, "let theaterChaseRainbow = 
function(wait) {
    let j = 0, q, i, p; 
    while(j < 256) {
        q = 0; 
        while(q < 3) {
            i = 0; 
            while(i < 8) {
                p = (i + j) % 255;
                setPixelColor(i + q, WheelR(p), WheelG(p), WheelB(p)); 
                i = i + 3;
            }
            show();
            delay(wait);
            i = 0;
            while(i < 8) {
                setPixelColor(i + q, 0, 0, 0);
                i = i + 3;
            }
            q++;
        }
        j++;
    }
    return j;
};

", 0);
```

**WORKING**
```
v = js_eval(js, "let theaterChaseRainbow = 
function(wait) {
    let j = 0, q, i, p, r; 
    while(j < 256) {
        q = 0; 
        while(q < 3) {
            i = 0; 
            while(i < 8) {
                p = (i + j) % 255;
                r = i + q;
                setPixelColor(r, WheelR(p), WheelG(p), WheelB(p)); 
                i = i + 3;
            }
            show();
            delay(wait);
            i = 0;
            while(i < 8) {
                r = i + q;
                setPixelColor(r, 0, 0, 0);
                i = i + 3;
            }
            q++;
        }
        j++;
    }
    return j;
};

", 0);
```

cpq, 2019-12-21:
Is it still an issue?

sfranzyshen, 2019-12-22:
seams to be working as expected in 0.0.19 ... TY

closed 2019-12-22 by sfranzyshen