cesanta/elk #17: minimal example

sfranzyshen, 2019-12-19:
Since that is a different issue, please open a new one - I'd appreciate a minimal example.
_Originally posted by @cpq in https://github.com/cesanta/elk/issues/14#issuecomment-567382091_

sfranzyshen, 2019-12-19:
**ESP8266** Tested Working
Sketch uses 276820 bytes (26%) of program storage space. Maximum is 1044464 bytes. Global variables use 28844 bytes (35%) of dynamic memory, leaving 53076 bytes for local variables. Maximum is 81920 bytes.

**ESP32** Tested Working
Sketch uses 234549 bytes (17%) of program storage space. Maximum is 1310720 bytes. Global variables use 15260 bytes (4%) of dynamic memory, leaving 312420 bytes for local variables. Maximum is 327680 bytes.

**UNO** Tested Not Working!!
Sketch uses 30528 bytes (94%) of program storage space. Maximum is 32256 bytes. Global variables use 2059 bytes (100%) of dynamic memory, leaving -11 bytes for local variables. Maximum is 2048 bytes. Not enough memory;

###After remarking out the theaterChaseRainbow() function it squeaks by the compile.

Sketch uses 30100 bytes (93%) of program storage space. Maximum is 32256 bytes. Global variables use 1671 bytes (81%) of dynamic memory, leaving 377 bytes for local variables. Maximum is 2048 bytes. Low memory available, stability problems may occur.

###Removed everything except the colorWipe() function ... Still nothing :(

Sketch uses 29140 bytes (90%) of program storage space. Maximum is 32256 bytes. Global variables use 873 bytes (42%) of dynamic memory, leaving 1175 bytes for local variables. Maximum is 2048 bytes.

**Minimal Example**
```
#include <Adafruit_NeoPixel.h>
#include <elk.h> 

#define ALLOC     1024 // tried as low as 600 for uno ...
#define LED_PIN   D2 // 6 for uno, D2 esp8266, 18 esp32
#define LED_COUNT 8

struct  js *js = js_create(malloc(ALLOC), ALLOC);
Adafruit_NeoPixel strip(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  js_import(js, "delay", (uintptr_t) Sys_delay, "vi");
  js_import(js, "show", (uintptr_t) Neo_show, "v");
  js_import(js, "clear", (uintptr_t) Neo_clear, "v");
  js_import(js, "setPixelColor", (uintptr_t) Neo_setPixelColor, "viiii");
  js_import(js, "WheelR", (uintptr_t) Neo_WheelR, "ii");
  js_import(js, "WheelG", (uintptr_t) Neo_WheelG, "ii");
  js_import(js, "WheelB", (uintptr_t) Neo_WheelB, "ii");
  js_import(js, "numPixels", (uintptr_t) Neo_numPixels, "i");

  js_eval(js, "let colorWipe = function(d, r, g, b) { let i = 0; while (i < numPixels()) { setPixelColor(i, r, g, b); show(); delay(d); i++; } delay(d * numPixels()); return i; };", 0);
  js_eval(js, "let theaterChase = function(d, r, g, bl) { let a = 0, b, c; while(a < 10) { b = 0; while(b < 3) { clear(); c = b; while(c < numPixels()) { setPixelColor(c, r, g, bl); c += 3; } show(); delay(d); b++; } a++; } return a; };", 0);
  js_eval(js, "let theaterChaseRainbow = function(wait) { let j = 0, q, i, p, r; while(j < 256) { q = 0; while(q < 3) { i = 0; while(i < numPixels()) { p = (i + j) % 255; r = i + q; setPixelColor(r, WheelR(p), WheelG(p), WheelB(p)); i = i + 3; } show(); delay(wait); i = 0; while(i < numPixels()) { r = i + q; setPixelColor(r, 0, 0, 0); i = i + 3; } q++; } j++; } return j; }; ", 0);
  js_eval(js, "let rainbowCycle = function(wait) { let j = 0, i, p; while(j < 256 * 5) { i = 0; while(i < numPixels()) { p = ((i * 256 / numPixels()) + j) & 255; setPixelColor(i, WheelR(p), WheelG(p), WheelB(p)); i++; } show(); delay(wait); j++; } return j; };", 0);
  js_eval(js, "let rainbow = function(d) { let j = 0, i, p; while(j < 256) { i = 0; while(i < numPixels()) { p = (i + j) & 255; setPixelColor(i, WheelR(p), WheelG(p), WheelB(p)); i++; } show(); delay(d); j++; } return j; };", 0);
   
  strip.begin();
  strip.show();
  strip.setBrightness(50);
}

extern "C" void Neo_show() { strip.show(); }
extern "C" void Neo_clear() { strip.clear(); }
extern "C" void Neo_setPixelColor(uint16_t n, int r, int g, int b) { strip.setPixelColor(n, strip.Color(r, g, b)); }
extern "C" void Sys_delay(unsigned long milli) { delay(milli); }
extern "C" int Neo_WheelR(int wheelpos) { return WheelR(wheelpos); }
extern "C" int Neo_WheelG(int wheelpos) { return WheelG(wheelpos); }
extern "C" int Neo_WheelB(int wheelpos) { return WheelB(wheelpos); }
extern "C" int Neo_numPixels(void) { return strip.numPixels(); }

int WheelR(int WheelPos) {
  WheelPos = 255 - WheelPos;
  if(WheelPos < 85) {
    return (255 - WheelPos * 3);
  }
  if(WheelPos < 170) {
    return (0);
  }
  WheelPos -= 170;
  return (WheelPos * 3);
}

int WheelG(int WheelPos) {
  WheelPos = 255 - WheelPos;
  if(WheelPos < 85) {
    return (0);
  }
  if(WheelPos < 170) {
    WheelPos -= 85;
    return (WheelPos * 3);
  }
  WheelPos -= 170;
  return (255 - WheelPos * 3);
}

int WheelB(int WheelPos) {
  WheelPos = 255 - WheelPos;
  if(WheelPos < 85) {
    return (WheelPos * 3);
  }
  if(WheelPos < 170) {
    WheelPos -= 85;
    return (255 - WheelPos * 3);
  }
  return (0);
}

void loop() {
  js_eval(js, "colorWipe(50, 255, 0, 0);", 0);
  js_eval(js, "colorWipe(50, 0, 255, 0);", 0);
  js_eval(js, "colorWipe(50, 0, 0, 255);", 0);
  js_eval(js, "theaterChase(50, 127, 127, 127);", 0);
  js_eval(js, "theaterChase(50, 127, 0, 0);", 0);
  js_eval(js, "theaterChase(50, 0, 0, 127);", 0);
  js_eval(js, "rainbow(30);", 0);
  js_eval(js, "rainbowCycle(20);", 0);  
  js_eval(js, "theaterChaseRainbow(50);", 0);
}
```

sfranzyshen, 2019-12-20:
or this one where the javascript functions are stored on the spiffs filesystem ... 

Sketch uses 307904 bytes (29%) of program storage space. Maximum is 1044464 bytes. Global variables use 27780 bytes (33%) of dynamic memory, leaving 54140 bytes for local variables. Maximum is 81920 bytes.

**SPIFFS**
```
#include <Adafruit_NeoPixel.h>
#include <elk.h> 
#include <FS.h>

#define ALLOC     2048
#define LED_PIN   D2
#define LED_COUNT 8
#define DEBUG     0

/* contents of init.js file
let colorWipe = function(d, r, g, b) { let i = 0; while (i < numPixels()) { setPixelColor(i, r, g, b); show(); delay(d); i++; } delay(d * numPixels()); };
let theaterChase = function(d, r, g, bl) { let a = 0, b, c; while(a < 10) { b = 0; while(b < 3) { clear(); c = b; while(c < numPixels()) { setPixelColor(c, r, g, bl); c += 3; } show(); delay(d); b++; } a++; } };
let theaterChaseRainbow = function(d) { let j = 0, q, i, p, r; while(j < 256) { q = 0; while(q < 3) { i = 0; while(i < numPixels()) { p = (i + j) % 255; r = i + q; setPixelColor(r, WheelR(p), WheelG(p), WheelB(p)); i = i + 3; } show(); delay(d); i = 0; while(i < numPixels()) { r = i + q; setPixelColor(r, 0, 0, 0); i = i + 3; } q++; } j++; } };
let rainbowCycle = function(d) { let j = 0, i, p; while(j < 256 * 5) { i = 0; while(i < numPixels()) { p = ((i * 256 / numPixels()) + j) & 255; setPixelColor(i, WheelR(p), WheelG(p), WheelB(p)); i++; } show(); delay(d); j++; } };
let rainbow = function(d) { let j = 0, i, p; while(j < 256) { i = 0; while(i < numPixels()) { p = (i + j) & 255; setPixelColor(i, WheelR(p), WheelG(p), WheelB(p)); i++; } show(); delay(d); j++; } };
let loop = function(d) { colorWipe(50, 255, 0, 0); colorWipe(50, 0, 255, 0); colorWipe(50, 0, 0, 255); theaterChase(50, 127, 127, 127); theaterChase(50, 127, 0, 0); theaterChase(50, 0, 0, 127); rainbow(30); rainbowCycle(20); theaterChaseRainbow(50); };
*/

const char* filename = "/init.js";

struct  js *js = js_create(malloc(ALLOC), ALLOC);
Adafruit_NeoPixel strip(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
  jsval_t v;
  char buffer[512];
  int l;
  
  Serial.begin(115200);
  Serial.println();
  
  if(DEBUG) {
    Serial.println("Starting ...");
  }

  js_import(js, "delay", (uintptr_t) Sys_delay, "vi");
  js_import(js, "print", (uintptr_t) Sys_print, "vj");
  js_import(js, "show", (uintptr_t) Neo_show, "v");
  js_import(js, "clear", (uintptr_t) Neo_clear, "v");
  js_import(js, "setPixelColor", (uintptr_t) Neo_setPixelColor, "viiii");
  js_import(js, "WheelR", (uintptr_t) Neo_WheelR, "ii");
  js_import(js, "WheelG", (uintptr_t) Neo_WheelG, "ii");
  js_import(js, "WheelB", (uintptr_t) Neo_WheelB, "ii");
  js_import(js, "jsinfo", (uintptr_t) js_info, "sm");
  js_import(js, "numPixels", (uintptr_t) Neo_numPixels, "i");

  //Initialize File System
  if(SPIFFS.begin())
  {
    if(DEBUG) { Serial.println("SPIFFS Initialize....ok"); }
    File f = SPIFFS.open(filename, "r");
    if (!f) {
      if(DEBUG) { Serial.println("file open failed"); }
    } 
    else
    {
      if(DEBUG) { Serial.println("Reading Data from File:"); }
      while (f.available()) {
        l = f.readBytesUntil('\n', buffer, sizeof(buffer));
        buffer[l] = 0;
        v = js_eval(js, buffer, 0);
        if(DEBUG) { Serial.print("js_eval() result: "); Serial.println(js_str(js, v)); }
        js_gc(js, v);
        //Serial.println(buffer);
      }      
      f.close();  //Close file
      if(DEBUG) { Serial.println("File Closed"); }
    }
  }
  else
  {
    if(DEBUG) { Serial.println("SPIFFS Initialization ... failed"); }
  }
     
  strip.begin();
  strip.show();
  strip.setBrightness(50);

  if(DEBUG) {
    v = js_eval(js, "print('VM Ready ...'); print(jsinfo(null));", 0);
    Serial.print("jsinfo() result: "); Serial.println(js_str(js, v));
    js_gc(js, v);
    Serial.println("Setup Complete ...");
  }
}

extern "C" void Neo_show() { strip.show(); }
extern "C" void Neo_clear() { strip.clear(); }
extern "C" void Neo_setPixelColor(uint16_t n, int r, int g, int b) { strip.setPixelColor(n, strip.Color(r, g, b)); }
extern "C" void Sys_delay(unsigned long milli) { delay(milli); }
extern "C" void Sys_print(jsval_t v) { Serial.println(js_str(js, v)); }
extern "C" int Neo_WheelR(int wheelpos) { return WheelR(wheelpos); }
extern "C" int Neo_WheelG(int wheelpos) { return WheelG(wheelpos); }
extern "C" int Neo_WheelB(int wheelpos) { return WheelB(wheelpos); }
extern "C" int Neo_numPixels(void) { return strip.numPixels(); }

int WheelR(int WheelPos) {
  WheelPos = 255 - WheelPos;
  if(WheelPos < 85) {
    return (255 - WheelPos * 3);
  }
  if(WheelPos < 170) {
    return (0);
  }
  WheelPos -= 170;
  return (WheelPos * 3);
}

int WheelG(int WheelPos) {
  WheelPos = 255 - WheelPos;
  if(WheelPos < 85) {
    return (0);
  }
  if(WheelPos < 170) {
    WheelPos -= 85;
    return (WheelPos * 3);
  }
  WheelPos -= 170;
  return (255 - WheelPos * 3);
}

int WheelB(int WheelPos) {
  WheelPos = 255 - WheelPos;
  if(WheelPos < 85) {
    return (WheelPos * 3);
  }
  if(WheelPos < 170) {
    WheelPos -= 85;
    return (255 - WheelPos * 3);
  }
  return (0);
}

void loop() {
  jsval_t v;
  
  if(DEBUG) { Serial.println("Looping ..."); }

  v = js_eval(js, "loop();", 0);
  if(DEBUG) { Serial.print("loop result: "); Serial.println(js_str(js, v)); }
  js_gc(js, v);
}
```

cpq, 2019-12-20:
That example is far from minimal :)
The minimal one is this:

```bash
$ cc -o elk examples/posix/main.c -I src -L src/linux-x64 -ldl -lelk
$ ./elk -e 'let f=function(x,y){return {x:x,y:y}};f(1+2,3*4)'
{"x":1,"y":14}
```
Released 0.0.19 with a fix - please verify.

closed 2019-12-20 by sfranzyshen