cesanta/elk #18: js_destroy()

sfranzyshen, 2019-12-22:
unlike with the predecessor (mjs) there is no destroy function ... is there another to free and restart the vm without rebooting? can I simply use free() ??

**create**
struct  js *js = js_create(malloc(ALLOC), ALLOC);
**destroy**
free(js);

cpq, 2019-12-23:
Elk uses the memory chunk you give to it, and it does not know how that memory chunk was created. Thus it does not have the destroy function. Example:

```c
char mem[500];
struct js *js = js_create(mem, sizeof(mem));
```

If you want to re-initialise Elk in the same memory chunk, just call `js_create()` again.

Beware the fact that some C functions that Elk can call, can create side-effects. For example, if you open some descriptor, you probably want to close it before deinitialising the engine. That what https://mdash.net/docs/userguide/elk.md firmware does - before calling `js_create()` again, it calls `deinit()` JS function, thus letting JS user to register deallocation logic to avoid leaks.

sfranzyshen, 2019-12-23:
> Beware the fact that some C functions that Elk can call, can create side-effects. For example, if you open some descriptor, you probably want to close it before deinitialising the engine. That what [...]

TY again! also I noticed a inaccuracy in https://mdash.net/docs/userguide/elk.md document ... 
in the section 7. Provision your board to mdash.net: 

`mos dash.token=COPIED_DEVICE_PASSWORD --port SERIAL_PORT
`
should be ... ??
`mos config-set dash.token=COPIED_DEVICE_PASSWORD --port SERIAL_PORT
`

closed 2019-12-23 by sfranzyshen