cesanta/elk #19: 'equal' & 'not equal' is not working ...

sfranzyshen, 2019-12-23:
LINUX_DIST="Linux Mint 18.3 Sylvia MATE 32-bit"
KERNEL=Linux computer 4.10.0-38-generic i686 i686 i686 GNU/Linux

ARDUINO IDE: 1.8.10 (32bit)
elk Library: version 0.0.19 (Installed using Arduino IDE library manager)
ESP32 boards (DEVKIT): 1.0.4 (Installed with board manager from Arduino IDE)

From the [elk github page](https://github.com/cesanta/elk#supported-standard-operations-and-constructs) ... it reads 

> All but !=, ==. Use !==, === instead

but the following code demonstrates that '===' isn't working (also tested '==' with same result)

**code**
```
    v = js_eval(js, "if(1 === 1) { print('working ...'); return 1; } print('not working ...'); return 0;", 0);
    Serial.print("test result: "); Serial.println(js_str(js, v));
    js_gc(js, v);
```

**result**
`test result: ERROR 1898
`
if I replace '===' with either the '>=' or '<=' it works as expected ...

**result**
```
"working ..."
test result: 1
```

sfranzyshen, 2019-12-23:
UPDATE: In trying to find a solution to not having an 'equal' available ... I stumbled on this problem also ... when a value is being compared with >= or <= it errors-out when the value is greater than the one it's being compared to ... it's easier to see the arising pattern from the results than to try and explain it ... 
  
**CODE**
```
let patCu = 0; if(patCu >= 2 && patCu <= 2) { print('working ...'); return 1; } print('not working ...'); return 0;
testing result: ERROR 1633

let patCu = 1; if(patCu >= 2 && patCu <= 2) { print('working ...'); return 1; } print('not working ...'); return 0;
testing result: ERROR 1633

let patCu = 2; if(patCu >= 2 && patCu <= 2) { print('working ...'); return 1; } print('not working ...'); return 0;
"working ..."
testing result: 1

let patCu = 3; if(patCu >= 2 && patCu <= 2) { print('working ...'); return 1; } print('not working ...'); return 0;
"not working ..."
testing result: 0

let patCu = 3; if(patCu <= 2 && patCu >= 2) { print('working ...'); return 1; } print('not working ...'); return 0;
testing result: ERROR 1633

let patCu = 2; if(patCu <= 2 && patCu >= 2) { print('working ...'); return 1; } print('not working ...'); return 0;
"working ..."
testing result: 1

let patCu = 1; if(patCu <= 2 && patCu >= 2) { print('working ...'); return 1; } print('not working ...'); return 0;
"not working ..."
testing result: 0

let patCu = 0; if(patCu <= 2 && patCu >= 2) { print('working ...'); return 1; } print('not working ...'); return 0;
"not working ..."
testing result: 0
```

sfranzyshen, 2019-12-24:
```
let patCu = 3; if(patCu >= 2) { print('working ...'); return 1; } print('not working ...'); return 0;
"working ..."
testing result: 1

let patCu = 3; if(patCu <= 2) { print('working ...'); return 1; } print('not working ...'); return 0;
"not working ..."
testing result: 0

let patCu = 3; if(patCu >= 2 && patCu <= 2) { print('working ...'); return 1; } print('not working ...'); return 0;
"not working ..."
testing result: 0

let patCu = 3; if(patCu <= 2 && patCu >= 2) { print('working ...'); return 1; } print('not working ...'); return 0;
testing result: ERROR 1633
```

cpq, 2019-12-24:
`if(1 === 1) { print('working ...'); return 1; }` is not a valid JavaScript snippet: return must be only from a function.

sfranzyshen, 2019-12-24:
> `if(1 === 1) { print('working ...'); return 1; }` is not a valid JavaScript snippet: return must be only from a function.

... but I'm still having issues with 'equal' & 'not equal'  not working ...

**Snippet**
```
    v = js_eval(js, "if(1 === 1) print('working ...');", 0);
    Serial.print("result: "); Serial.println(js_str(js, v));
    js_gc(js, v);
```

**Result**
`result: ERROR 1898
`

**Snippet**
```
    v = js_eval(js, "let a=1, b=2; if(a !== b) print('working ...');", 0);
    Serial.print("result: "); Serial.println(js_str(js, v));
    js_gc(js, v);
```

**Result**
`result: ERROR 1898
`

cpq, 2019-12-24:
Sure, there is an issue. But please use valid constructs in tests. Thanks!

cpq, 2019-12-24:
Released 0.0.20

sfranzyshen, 2019-12-25:
> Released 0.0.20

that update rendered my working code to not working code ... I'll look at it tomorrow

cpq, 2019-12-25:
Every fix has an associated unit tests added, thus each release must not break what was working.
If it breaks though, that means there are some more issues not yet covered by unit tests.
As always, I'd appreciate a snippet to reproduce.

sfranzyshen, 2019-12-25:
so the problem was the number of variables being set with a single let statement ... seams that number got smaller.

**Not Working**
`let pixPr = 0, patPr = 0, patCu = 0, patIn = 5500, pixIn = 45, pixQu = 0, pixCy = 0, pixCu = 0, pixNu = numPixels();`

**Working**
`let patPr = 0, patCu = 0; let patIn = 5500, pixPr = 0, pixIn = 45, pixQu = 0, pixCy = 0, pixCu = 0, pixNu = numPixels();`

Also I can confirm that the equal '===' is indeed functioning as expected :) TY!

closed 2019-12-25 by sfranzyshen