cesanta/elk #20: use of return from within a function causes irregular results ...

sfranzyshen, 2019-12-24:
LINUX_DIST="Linux Mint 18.3 Sylvia MATE 32-bit"
KERNEL=Linux computer 4.10.0-38-generic i686 i686 i686 GNU/Linux

ARDUINO IDE: 1.8.10 (32bit)
elk Library: version 0.0.19 (Installed using Arduino IDE library manager)
ESP32 boards (DEVKIT): 1.0.4 (Installed with board manager from Arduino IDE)

While trying to work around the non-functioning 'equal' operator I encountered strange unexpected result from calling subsequent function calls (not executing) after using 'return' ... here again are  comparisons of  'working' and 'not working'  code snippets ...

**Not Working**
```
let loop = function() {
  let curMi = millis();
  if(curMi - patPr >= patIn) {
    patPr = curMi;
    patCu++;
    if(patCu >= 9) patCu = 0;
  }
  if(curMi - pixPr >= pixIn) {
    pixPr = curMi;
    if(patCu >=8) {tCR(); return;}
    if(patCu >=7) {rbC(); return;}
    if(patCu >=6) {rb(); return;}
    if(patCu >=5) {tC(0, 0, 127); return;}
    if(patCu >=4) {tC(127, 0, 0); return;}
    if(patCu >=3) {tC(127, 127, 127); return;}
    if(patCu >=2) {cW(0, 0, 255); return;}
    if(patCu >=1) {cW(0, 255, 0); return;}
    if(patCu >=0) {cW(255, 0, 0); return;}
  }    
};
```

**Working**
```
let loop = function() {
  let curMi = millis();
  if(curMi - patPr >= patIn) {
    patPr = curMi;
    patCu++;
    if(patCu >= 9) patCu = 0;
  }
  if(curMi - pixPr >= pixIn) {
    pixPr = curMi;
    if(patCu >=8) {if(patCu <=8) tCR();}
    if(patCu >=7) {if(patCu <=7) rbC();}
    if(patCu >=6) {if(patCu <=6) rb();}
    if(patCu >=5) {if(patCu <=5) tC(0, 0, 127);}
    if(patCu >=4) {if(patCu <=4) tC(127, 0, 0);}
    if(patCu >=3) {if(patCu <=3) tC(127, 127, 127);}
    if(patCu >=2) {if(patCu <=2) cW(0, 0, 255);}
    if(patCu >=1) {if(patCu <=1) cW(0, 255, 0);}
    if(patCu >=0) {if(patCu <=0) cW(255, 0, 0);}
  }    
};
```

sfranzyshen, 2019-12-26:
0.0.22 throws a wdt reset ...

cpq, 2021-05-01:
Shoud be fixed now.

closed 2021-05-01 by cpq