cesanta/elk #25: WebAssembly

sfranzyshen, 2020-02-13:
Would you consider releasing a WebAssembly version of elk? For the purpose of testing Elk code from within a web based editor.

cpq, 2020-03-03:
Elk does not have an intermediate byte code compilation stack.
It executes source code in-place. So it cannot generate a WebAssembly output.
Is that what you were asking for?

sfranzyshen, 2020-03-03:
> Elk does not have an intermediate byte code compilation stack.
> It executes source code in-place. So it cannot generate a WebAssembly output.
> Is that what you were asking for?

No, I was asking if a version of the elk interpreter could be released as a wasm binary so that it could be used from within the browser to validate (elk) code. But, as your site points out elk is a subset of ES6 ... so i'm going to use a seperate linter like JSHint, ESLint, ...

closed 2020-03-03 by sfranzyshen