cesanta/elk #28: No longer compiles on esp32

H4ckd4ddy, 2021-03-22:
Hi 👋,

First thanks for this awesome lib :)

Unfortunately, it no longer seems functional on esp32. Impossible to compile, even the example provided

I tried to go back to old versions of arduino and the software of the esp32 board because it worked well before, but without success. I can't figure out what has changed since my last use of elk.

Here is the error output :
`
/Users/esellan/Documents/Arduino/libraries/elk/src/esp32/libelk.a: In function `js_gc':
elk.c:(.text+0x7d0): multiple definition of `js_gc'
/Users/esellan/Documents/Arduino/libraries/elk/src/esp32/libelk.a:elk.c:(.text+0x7d0): first defined here
/Users/esellan/Documents/Arduino/libraries/elk/src/esp32/libelk.a: In function `js_fmt':
elk.c:(.text+0x139c): multiple definition of `js_fmt'
/Users/esellan/Documents/Arduino/libraries/elk/src/esp32/libelk.a:elk.c:(.text+0x139c): first defined here
/Users/esellan/Documents/Arduino/libraries/elk/src/esp32/libelk.a: In function `js_str':
elk.c:(.text+0x15d8): multiple definition of `js_str'
/Users/esellan/Documents/Arduino/libraries/elk/src/esp32/libelk.a:elk.c:(.text+0x15d8): first defined here
/Users/esellan/Documents/Arduino/libraries/elk/src/esp32/libelk.a: In function `js_eval':
elk.c:(.text+0x1604): multiple definition of `js_eval'
/Users/esellan/Documents/Arduino/libraries/elk/src/esp32/libelk.a:elk.c:(.text+0x1604): first defined here
/Users/esellan/Documents/Arduino/libraries/elk/src/esp32/libelk.a: In function `js_parse':
elk.c:(.text+0x2300): multiple definition of `js_parse'
/Users/esellan/Documents/Arduino/libraries/elk/src/esp32/libelk.a:elk.c:(.text+0x2300): first defined here
/Users/esellan/Documents/Arduino/libraries/elk/src/esp32/libelk.a: In function `js_ffi':
elk.c:(.text+0x2a1c): multiple definition of `js_ffi'
/Users/esellan/Documents/Arduino/libraries/elk/src/esp32/libelk.a:elk.c:(.text+0x2a1c): first defined here
/Users/esellan/Documents/Arduino/libraries/elk/src/esp32/libelk.a: In function `js_import':
elk.c:(.text+0x2a70): multiple definition of `js_import'
/Users/esellan/Documents/Arduino/libraries/elk/src/esp32/libelk.a:elk.c:(.text+0x2a70): first defined here
/Users/esellan/Documents/Arduino/libraries/elk/src/esp32/libelk.a: In function `js_info':
elk.c:(.text+0x2ac0): multiple definition of `js_info'
/Users/esellan/Documents/Arduino/libraries/elk/src/esp32/libelk.a:elk.c:(.text+0x2ac0): first defined here
/Users/esellan/Documents/Arduino/libraries/elk/src/esp32/libelk.a: In function `js_create':
elk.c:(.text+0x2af8): multiple definition of `js_create'
/Users/esellan/Documents/Arduino/libraries/elk/src/esp32/libelk.a:elk.c:(.text+0x2af8): first defined here
collect2: error: ld returned 1 exit status
exit status 1
Erreur de compilation pour la carte ESP32 Dev Module
`

Let me know if you need any information about this issue
Thanks by advance
best regards

cpq, 2021-04-29:
@H4ckd4ddy thank you.

We'll open source this soon, so compilation would not be a problem.
Could you elaborate on your use case please?

H4ckd4ddy, 2021-04-29:
Hi @cpq, thanks for your response.
We actually develop a student project, and we use this library to execute JS from an SD card connected to our esp32.
Thanks by advance,
Best regards

cpq, 2021-05-01:
Sources opened.

closed 2021-05-01 by cpq

H4ckd4ddy, 2021-05-02:
Thank you so much 🤩