cesanta/elk #30: the logical NOT (!) operator is not working ...

sfranzyshen, 2021-05-21:
I am unable to use the logical NOT (!) operator ... looking at your source code from the unit_test.c the following test fail ...

```
assert(ev(js, "a=true;a", "true"));
assert(ev(js, "a=!a;a", "false"));
assert(ev(js, "!123", "false"));
assert(ev(js, "!0", "true"));

```
here's what I get when using the cli ...

```
js > a=true;a
true
js > a=!a;a
ERROR: type mismatch
js > a=!a;a
ERROR: type mismatch
js > !123
ERROR: unknown op 104
js > !0
ERROR: unknown op 104
js > a=!(a);a
ERROR: type mismatch
js > !(a);
ERROR: type mismatch
js > a;
true
```

cpq, 2021-05-22:
Your unit test is integrated and works with no code change. Please double-check that you're using the latest code.

Repl words as expected, too:

```console
lsm:~/src/elk/test$ make cli
cc ../elk.c ../examples/cmdline/main.c -W -Wall -Werror -Wno-deprecated -I.. -g -fsanitize=address,undefined -coverage  -o cli
lsm:~/src/elk/test$ .//cli 'let a=true;a=!a;a'
false
Executed in 0.299 ms. Mem usage is 0% of 8192 bytes.
lsm:~/src/elk/test$ .//cli '!123'
false
Executed in 0.224 ms. Mem usage is 0% of 8192 bytes.
lsm:~/src/elk/test$ .//cli '!0'
true
Executed in 0.223 ms. Mem usage is 0% of 8192 bytes.
```

closed 2021-05-22 by cpq

sfranzyshen, 2021-05-22:
sorry that was my problem ... thank you

commit f07cc356 (2021-05-22): Fix #30