cesanta/elk #40: string end with random character

Lucy2003, 2021-08-16:
Hi,
I'm trying to use this library on ESP8266 (Core 2.6.3,with Arduino IDE 1.8.12).

I added a C function "print" to JS,and found that: When calling this C function from JS with string as a parameter continuously,random characters appear at the end of the string.Even when calling not the same function continuously,random character will appear in the output from serial.But the first call outputed normal.That's was quiet strange.

My sketch files are here: [BlinkyJS-lucy2003.zip](https://github.com/cesanta/elk/files/6989994/BlinkyJS-lucy2003.zip)
Note that I load scripts from SPIFFS,which means scripts will need to be uploaded by 'ESP8266 Sketch Data Upload' tool.Otherwise it will need to be wrote in the code.

Any solution?Thanks you.

cpq, 2021-08-25:
@Lucy2003 thanks

Could you provide an example output please?

Lucy2003, 2021-08-25:
Sorry,I forgot to provide an example output ;)
After uploading my sketch file and data files provided above to ESP8266,it outputed in Serial Monitor:

```

=== Elk JavaScript runtime on ESP8266 ===

[RT] Loading file: '/main.js' - 142 bytes done.Executing...

[RT] Loading file: '/blink.js' - 269 bytes done.Executing...

[RT] Loading file: '/blink2.js' - 251 bytes done.Executing...
Hello,JS on ESP8266!4
Here we are going to blink!
Blink1t
Blink2

```

Expected output:
```
(Debug output here)

Hello,JS on ESP8266!
Here we are going to blink!
Blink1
Blink2

```

cpq, 2021-09-01:
@Lucy2003 verified on ESP32 using Mongoose OS, no trailing garbage gets printed..

What if you add modify a loop this way - would the hardcoded print() call also produce garbage?

```c
void loop() {
    Serial.printf("\n[RT] Warning: No more code to run.Runtime reset.");
    initJS();
    runJS("/main.js");
    js_eval(js, "print('hi!\n');", 16);   // <-- add this line
}
```

closed 2021-09-21 by cpq

Lucy2003, 2021-09-21:
Oh,it does not produce garbage output but I don't know why...

cpq, 2021-09-21:
@Lucy2003 the fix was pushed recently. Courtesy of @leonzhu211

commit ffb29e7b (2021-09-21): Fix #40, fix #41 - proper 0-terminate in mkentity()