cesanta/elk #43: Why compaction instead of regular mark&sweep?

playXE, 2021-09-26:
GC right now seems to compact heap but it does compact it in a really slow way and as far as I understood it walks the entire heap when it changes a single reference. Maybe it is worth having mark&sweep with a linked list allocator instead of such a slow GC algorithm?

cpq, 2021-09-27:
@playXE you're correct, a compaction is run on each GC run - that's the simplest way to keep memory defragmented. Agree that   optimisation is possible. It is all matter of implementation: here, simplicity is first, performance is second. GC on itself is complex enough to understand. If you have ideas / examples on efficient optimisation, please speak up.

playXE, 2021-09-27:
Yeah I agree that for such small engine simplicity should be first, but if it is going to be used on micro controllers wasting too much time imo to just perform compaction is a bad thing so what I think can be done is that:
1) Separate heap into chunks of N bytes size (maybe choose when compiling elk?) from 256 bytes to maybe a few kilobytes, I assume elk objects are quite small
2) On each GC choose the most unfragmented/empty chunks and copy all objects to these chunks from fragmented blocks and chunks that are empty could be reclaimed to use by OS or could be used for allocation

This algorithm sounds quite like Immix so I guess at this point even Immix could be implemented? The idea of Immix is pretty simple: instead of operating on blocks, it operates on lines. So let's say we have 1KB memory block and this block is separated into 8 lines each of these lines is 128 bytes in size and how it would work: when allocating search through bitmap for first unmarked and first marked lines, that is a "hole" we will allocate into, if an object is too large then the object may split two lines safely. When collecting garbage you do not have to mark objects separately instead you just mark *lines* where an object is allocated and this allows you to entirely remove the sweep phase. When compacting Immix allows one to easily find fragmented blocks because a fragmented block is a block that has too many holes in it.

cpq, 2021-09-28:
Avoiding GC before each top-level statement, and deferring it until memory reaches some threshold, increases speed.

This commit https://github.com/cesanta/elk/commit/0ecf79dbfb46d25301a5312e87c23a3b32dfbfa6 introduces 75% threshold and speeds up commented-out test case with a loop from 3 sec

```
./ut
done in 3156.52 ms
```

down to 1ms:

```
./ut
done in 1.162 ms
```

On this particular case, it is a several orders of magnitude improvement. Makes it less deterministic but could be worth it.

closed 2021-09-28 by cpq