cesanta/elk #45: The public API should allow creating an object

maelp, 2021-09-29:
Hi,

I'd like to use elk and to allow JS code to access a global object representing the state of the system, but right now it seems the API does not allow to do that easily.

Basically I'd like to be able to define a global object with the state of my sensors, so that the JS code can observe it.

A possible way would be to do like in the README and define methods for all the properties, but this means they have to be re-evaluated each time, eg:

```
jsval_t state = js_mkobj(js);

jsval_t a_jsval_imported_method_to_getSensor1 = js_import(...);
js_set(js, state, "getSensor1", a_jsval_imported_method_to_getSensor1);

jsval_t a_jsval_imported_method_to_getSensor2 = js_import(...);
js_set(js, state, "getSensor2", a_jsval_imported_method_to_getSensor2);
```

another option would be to use `js_eval` to create on-the-fly the value, but this means we have to serialize the value to a string, and then have elk parse it again

```
jsval_t state = js_mkobj(js);
js_set(js, state, "getSensor1", js_eval(..., "0.5"));
js_set(js, state, "getSensor2", js_eval(..., "61"));
```

what I'd like is a simple way to build jsval_t from standard values exposed in the API to allow doing something like:

```
jsval_t state = js_mkobj(js);
js_set(js, state, "getSensor1", js_float(0.5));
js_set(js, state, "getSensor2", js_int(61));
```

cpq, 2021-09-30:
@maelp you're asking to add extra API `js_string(js, "foo")`, `js_number(js, 1.234)` ?

maelp, 2021-09-30:
That would be perfect :)

maelp, 2021-09-30:
Side question that I've asked here: https://github.com/cesanta/elk/discussions/44

on an ESP32, would you recommend `elk` or `mjs`, and what are the differences?

cpq, 2021-09-30:
Here we go - https://github.com/cesanta/elk/commit/8f21c40d70e04c577ef64bd147eded3bcad4f04a

On ESP32, use Elk.

closed 2021-09-30 by cpq