cesanta/elk #46: Cannot make simple program work

maelp, 2021-09-30:
I'm trying this code

```
print("what is the value?");
let obj = {sensor: 10.0}
if (obj.sensor >= 1) {
    print("greater");
} else {
    print("smaller");
};
```

which correctly shows the first "what is the value?" line, but then exits without showing either of the other print line, and I'm not sure why?

cpq, 2021-09-30:
Semicolon is missing on line 2.

maelp, 2021-09-30:
😅

maelp, 2021-09-30:
thanks it works, small other question I had: how are we supposed to call methods that take a `jsval_t` parameter? does `elk` convert any kind of object? For instance can I create a method to which I can send any kind of data like

```
print_any({sensorValue: 1, otherParam: 2});
```

cpq, 2021-09-30:
I think a practical approach would be to have an editor (or a plugin to an existing editor) that would hightlight things like that, and suggest / show / autocomplete available API.

A `jsval_t` parameter is marshalled using `j` specifier. Your next question would be - how to traverse JS object in C. The answer would be - there is no way to do it. Elk does not expect you to do things like that. Import C functions in a way they know nothing about JS.

maelp, 2021-09-30:
thanks!

my use-case would have been to simply call `js_str` on the `jsval_t` and do something with the string, but it seems I can't even create a FFI method to which I can send any JS value? would you have an example?

cpq, 2021-09-30:
Could you elaborate on that please - what are you trying to achieve on a high level?

maelp, 2021-09-30:
I want to possibly send data to my server through MQTT but in an agnostic way, so people could do

```
sendData({errorType: "sensorValue", sensorValue: state.getValue()});
```

or 

```
sendData({"errorType": "log", "msg": "Some message"});
```

and the `jsval_t` would be converted to JSON and sent to the server

cpq, 2021-09-30:
That is a very reasonable use case.

Why not importing `js_str()` ? See https://github.com/cesanta/elk/blob/2ac8e9948c6c4b91ffc99fbc92c088526372281b/test/unit_test.c#L484  how unit test does that.

cpq, 2021-09-30:
Also, an existing ESP32 example does exactly that, with MQTT and all that. Did you look at examples at all?
See https://github.com/cesanta/elk/blob/2ac8e9948c6c4b91ffc99fbc92c088526372281b/examples/Esp32JS/editor/editor.js#L23

maelp, 2021-09-30:
Interesting thanks!

https://github.com/cesanta/elk/blob/2ac8e9948c6c4b91ffc99fbc92c088526372281b/examples/Esp32JS/editor/editor.js#L23

<https://github.com/cesanta/elk/issues/46#issuecomment-931366693>, or
unsubscribe
or Android

closed 2021-09-30 by cpq