cesanta/elk #51: ERROR: bad sig when using space in function declaration

maxint-rd, 2022-02-12:
Defining this function gives no error (note the space in "function (x)"):
```let fnBadSpace=function (x) { return(x+3); }```

When calling this function ERROR: bad sig is reported:
```
fnBadSpace(10);
ERROR: bad sig
```

Note that when listing the function definition there is a space in the definition:
```
fnBadSpace(10);
" (x) { return(x+3); }"
```

Perhaps the parser should skip this space when defining the function?
(When testing the library it took me a while to discover that this space was the culprit)

cpq, 2022-02-13:
"bad sig"? That's FFI error.
Can you describe exactly what you're doing please and what do you mean by "calling this function"?

maxint-rd, 2022-02-13:
Hi, thanks for the swift reply. 

What I'm doing is described above. In my ESP I can submit a JS string in the serial console to be evaluated and the result is printed.

I first enter the function definition:
```let fnBadSpace=function (x) { return(x+3); }```
(FYI: The result is 'undefined'). Then enter the function call:
```fnBadSpace(10);```
The result is this error:
```ERROR: bad sig```

Note that there is a space between "function" and "(x)".

commit 884c13ed (2022-02-28): Fix #51 - func literal parsing
closed 2022-02-28 by cpq