cesanta/elk #52: Garbage collection doesn't fixup object references to upper and deletes used entities from memory.

maxint-rd, 2022-02-16:
Hello again.
While further testing Elk on my ESP project and after incorporation of my previous fix, I encountered another issue with the garbage collection. It only occurs in certain circumstances and produces unpredictable behavior.

In my project it was very reproducible and  failed every time I ran this minimized test script:
   ```  let fn=function(x) { let i=0; while(i<x) {res=res+"!"; i++; }; }; let res="123"; fn(3);```
In my ESP-sketch this script results in a WDT-reset. Different variants of this script often wouldn't show the issue the same way.
The proper functioning should be that global variable res is set to the value of "123!!!".

After lengthy debugging by inserting debug dumps and debug printfs in js_stmt() and in js_delete_marked_entities() I think I found the issue. Long story short it appears that js_fixup_offsets() doesn't fixup the offset to upper for objects. Having an incorrect reference in upper can result in incorrect deletions in subsequent garbage collections, giving further issues, During my debug sessions I saw the upper reference remained the same after in-between T_STR values for res were deleted by CG. It think this issue floated to the surface now because of the more frequent garbage collection.

After first trying to understand your code, by analyzing debug output and based on that adding comments, I managed to make the above script function work without generating a WDT reset. This is the modified version of js_fixup_offsets():

```
static void js_fixup_offsets(struct js *js, jsoff_t start, jsoff_t size) {
  // fixup all entities prior to moving data at start for deleting the start entity (which is marked for deletion) 
  for (jsoff_t n, v, off = 0; off < js->brk; off += n) {  // start from 0!
    v = loadoff(js, off);       // load 4-byte offset value of item at off. 
                                         // For objects v is only the offset of its first property. For properties it is the offset of the next item (0 if first)
    n = esize(v & ~GCMASK);
    if (v & GCMASK) continue;  // To be deleted, don't bother
    if ((v & 3) != T_OBJ && (v & 3) != T_PROP) continue;      // T_STR doesn't need further fixup

    // Fixup first/next offset for values beyond starting point 
    if (v > start) saveoff(js, off, v - size);      // save new offset (v-size) for first/next at off  
      
    // MMOLE 220216: proper fixup of upper offset for objects that don't fall under global scope (upper@0)
    if ((v & 3) == T_OBJ) {
      jsoff_t  u = loadoff(js, off+sizeof(jsoff_t));       // load 4-byte offset value of upper
      if(u > start)
        saveoff(js, off+sizeof(jsoff_t), u - size); // fixup upper, checked for negatives
    }
    
    // fixup property, key and value
    if ((v & 3) == T_PROP) {
      
      // fixup key offset
      jsoff_t koff = loadoff(js, off + sizeof(off));
      if (koff > start) saveoff(js, off + sizeof(off), koff - size);
      
      // fixup value data offset
      jsval_t val = loadval(js, off + sizeof(off) + sizeof(off));
      if (is_mem_entity(vtype(val)) && vdata(val) > start) {
        // printf("MV %u %lu -> %lu\n", off, vdata(val), vdata(val) - size);
        saveval(js, off + sizeof(off) + sizeof(off), mkval(vtype(val), vdata(val) - size));
      }
    }
  }

  // fixup callbacks
  for (jsoff_t i = 0; i < js->ncbs; i++) {
    jsoff_t base = js->size + i * 3 * sizeof(jsoff_t) + sizeof(jsoff_t);
    jsoff_t o1 = loadoff(js, base), o2 = loadoff(js, base + sizeof(o1));
    if (o1 > start) saveoff(js, base, o1 - size);
    if (o2 > start) saveoff(js, base + sizeof(jsoff_t), o2 - size);
  }
  // Fixup js->scope
  jsoff_t off = vdata(js->scope);
  if (off > start) js->scope = mkval(T_OBJ, off - size);
}
```

Disclaimer: Please note that I only tried to fix this issue as I encountered it. Using my code is at your own risk. No guaranties given and no liability accepted. Unfortunately I am not able to provide support....

maxint-rd, 2022-02-17:
As js_gc is called at every statement, this may affect currently executed functions.
When executing function code we can't simply  delete items that are in memory before the currently executed  code (eg. global variables), as that can mess up our token-position, resulting in the next token to be different.
For that reason we also need to fixup the js->pos and perhaps other references too, or take other measures to prevent moving the currently executed code.

This is the test script that failed:
```let res2="aaa"; let fn=function(x) { let i=0; while(i<x) {res2=res2+"!"; i++;}; }; fn(10);```
Note that this time, the variable that gets a longer value is located before the function definition.
At the first iteration the STR value is reallocated, making the old allocation candidate for GC.
If the current js->pos points to an affected function, it is no longer valid after the memory was moved.

cpq, 2022-05-30:
Could you make a PR with your change please?

cpq, 2022-06-08:
@maxint-rd sounds good, thank you.
Could you elaborate a little on your project please?

maxint-rd, 2022-06-08:
Sure... I've been looking for a light weight scripting engine to use in my collection of ESP8266 boards. I've made a few small (hobby) projects such as a clock and a robot cart that use the ESP. You can find a video of them on my Youtube channel.

I'm looking for a common setup to share among various projects, in which the ESP8266 features a FileManager and website, an MQTT connection and some project specific control and behavior. For ease of programming I thought Javascript could be useful, sharing some compatibility between the embedded website and the JS driven hardware. While keeping some features hardcoded in firmware, some others can then be changed on the whim by JS, possibly even distributing changed code via MQTT. 

I have looked at some alternatives such as MicroPython, Basic and WebAssembly but for now prefer Javascript if I can get it working according my wishes. Of course using a larger MCU (such as ESP32) would have been easier, but that will not put my current MCU collection to work.

While testing ELK I encountered the issues mentioned earlier and some more, as I tried to get more features working. While available memory is larger on the ESP than on most Atmels, stable networking requires a lot of free RAM. This limits the size of the JS memory area and also the size of available stack. Using PROGMEM allows for more features in less memory. (Even though any reasonable script still easily fills it). 

Hope this gives you a bit more context to my experiments. As you might have read, it's more about the challenge of limitations than about practicality and real world use.

pavvmagic, 2022-06-09:
With your fixes to the js_fixup_offsets and js_delete_marked_entities functions, my stress test worked:
`

    assert(ev(jsp,
      "let tmr1cb;os.tmr_init(1,function(){tmr1cb();},null);"
      "let tmr2cb;os.tmr_init(2,function(){tmr2cb();},null);"
      "let tmr3cb;os.tmr_init(3,function(){tmr3cb();},null);"
      "let tmr4cb;os.tmr_init(4,function(){tmr4cb();},null);"
      "let setTimeout1=function(fn,ms){tmr1cb=fn;os.tmr_start(1,ms,false);};"
      "let setTimeout2=function(fn,ms){tmr2cb=fn;os.tmr_start(2,ms,false);};"
      "let setTimeout3=function(fn,ms){tmr3cb=fn;os.tmr_start(3,ms,false);};"
      "let setTimeout4=function(fn,ms){tmr4cb=fn;os.tmr_start(4,ms,false);};"
      "let setInterval1=function(fn,ms){tmr1cb=fn;os.tmr_start(1,ms,true);};"
      "let setInterval2=function(fn,ms){tmr2cb=fn;os.tmr_start(2,ms,true);};"
      "let setInterval3=function(fn,ms){tmr3cb=fn;os.tmr_start(3,ms,true);};"
      "let setInterval4=function(fn,ms){tmr4cb=fn;os.tmr_start(4,ms,true);};"
      " setTimeout1(function(){ "
      "  os.dbgs('#1'); "
      "  setInterval1(function(){ "
      "    os.dbgs('&1'); "
      "  },100) "
      " },1000); "
      " setTimeout2(function(){ "
      "  os.dbgs('#2'); "
      "  setInterval2(function(){ "
      "    os.dbgs('&2'); "
      "  },150) "
      " },2000); "
      " setTimeout3(function(){ "
      "  os.dbgs('#3'); "
      "  setInterval3(function(){ "
      "    os.dbgs('&3'); "
      "  },125) "
      " },3000); "
      " setTimeout4(function(){ "
      "  os.dbgs('#4'); "
      "  setInterval4(function(){ "
      "    os.dbgs('&4'); "
      "  },115) "
      " },4000); "
      " 0", "0"));`

cpq, 2022-07-15:
@maxint-rd your change is integrated, thank you!