cesanta/elk #53: Question how to write on MCP23017

kaiquecav, 2022-04-04:
Hi there! I have been facing an issue with my code that I am unable to find out a solution (I don't think it is a issue with the library). I am using MCP23017 over I2C with my esp32 and I am unable to create a function that works to digitalWrite on MCP23017. Could someone help me? My code is the Esp32JS example with these add.

**This one is inside extern "C"**
`void mcp_write(int pin, bool val) {
  mcp1.digitalWrite(pin, val);
}`

**This new line below**
`js_set(js, js_glob(js), "gpio", gpio);

js_set(js, gpio, "mode", js_import(js, (uintptr_t) gpio_mode, "vii"));

js_set(js, gpio, "write", js_import(js, (uintptr_t) gpio_write, "vii"));

js_set(js, gpio, "read", js_import(js, (uintptr_t) gpio_read, "ii"));

js_set(js, gpio, "mcpwrite", js_import(js, (uintptr_t) mcp_write, "vib")); //my new mcp write`

**When I do this I get this error:**
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.
Core 1 register dump:
PC      : 0x4011b26b  PS      : 0x00060830  A0      : 0x8011b6e8  A1      : 0x3ffb1e10
A2      : 0x00000133  A3      : 0x3ffb1e9c  A4      : 0x00000032  A5      : 0x3ffbef4c
A6      : 0x00004e20  A7      : 0x60013000  A8      : 0x00000113  A9      : 0x3ffb1df0
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffb8548  A13     : 0x3ffcd8e4
A14     : 0x00002710  A15     : 0x3ffbef4c  SAR     : 0x0000001c  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace: 0x4011b26b:0x3ffb1e10 0x4011b6e5:0x3ffb1e40 0x4010f97d:0x3ffb1e60 0x4010fa31:0x3ffb1e90 0x4010fab1:0x3ffb1ec0 0x4010fb3d:0x3ffb1ee0 0x4010fc87:0x3ffb1f00 0x400d7c39:0x3ffb1f20 0x400ec36d:0x3ffb1f40 0x400f8d34:0x3ffb1f80 0x4011c38d:0x3ffb1fb0 0x40089911:0x3ffb1fd0

Rebooting...

kaiquecav, 2022-04-04:
So my main code has the adaifruit library.

`

#include <WiFi.h>
#include "JS.h"
#include <Adafruit_MCP23017.h>
Adafruit_MCP23017 mcp1;

const char *ssid = "WIFI_NETWORK";
const char *pass = "WIFI_PASSWORD";

void setup() {
  Serial.begin(115200);
  WiFi.begin(ssid, pass);
  while (WiFi.status() != WL_CONNECTED) Serial.print("."), delay(300);
  Serial.print("\nConnected, IP address: ");
  Serial.println(WiFi.localIP());

  mcp1.pinMode(8, OUTPUT);
  mcp1.pinMode(9, OUTPUT);
  mcp1.pinMode(10, OUTPUT);
  mcp1.pinMode(11, OUTPUT);
  mcp1.pinMode(12, OUTPUT);
  mcp1.pinMode(13, OUTPUT);
  mcp1.pinMode(14, OUTPUT);
  mcp1.pinMode(15, OUTPUT);

  JS.begin();
}

void loop() {
}

 `

maxint-rd, 2022-04-09:
Have you tried "vii" instead of "vib" (like gpio_write does),  to supply an integer instead of boolean?

My project uses the ESP8266 and I find the Exception Decoder useful to get more information on where the crash happens.
I think you need to find out where that divide by zero error occurs precisely.

cpq, 2022-05-13:
@kaiquecav use https://mdash.net/docs/#crash-backtrace to stringify the backtrace

cpq, 2022-05-30:
No response, closing this. Please reopen if you have your backtrace stringified.